pulp-platform/axi #1: axi_address_resolver relies on non-existing module

bchetwynd, 2018-09-06:
The axi_address_resolver component relies on the find_first_one component, which is not part of the repository.  The axi_address_resolver component is used in axi_lite_xbar.sv.

Thus, when attempting to simulate using compile_vsim.sh / run_vsim.sh scripts, the simulation fails due to undefined module.

FrancescoConti, 2018-09-06:
I'm not entirely sure what the local testbench does, however you can find all the IP dependencies in the `Bender.yml` file. In this case, what you need is http://github.com/pulp-platform/common_cells .

I think the IP owners (@fabianschuiki @zarubaf ?) should explicitly call `bender` to fetch these in their script.

bchetwynd, 2018-09-06:
Clarification much appreciated.

zarubaf, 2018-09-06:
Yes, @FrancescoConti is completely right. Currently we do not provide our `bender` flow as open-source. If you want to use the testbench you are welcome to do so. The find first zero can be found in the common cells repository as Francesco mentioned.

bchetwynd, 2018-09-06:
Thank you.  For what it is worth, the find_first_one module is contained in a deprecated directory.  However, including that file as suggested addresses the issue in the axi_lite_xbar.sv module.

gurkaynak, 2018-09-06:
@bchetwynd is that by any chance Brendon from WPI end of 90s?

fabianschuiki, 2018-09-08:
You can theoretically already use Bender, but you would have to compile it from source yourself (https://github.com/fabianschuiki/bender). We're still ironing out the flow of that tool and will provide precompiled binaries once things become more stable. But just in case you're feeling lucky 😉 .

closed 2018-10-23 by bchetwynd