pulp-platform/axi #2: axi_lite_xbar component arbitration

bchetwynd, 2018-10-17:
I am leveraging the axi_lite_xbar component to interface a hardware architecture processor to a variety of peripherals.  As I am not terribly concerned with performance, I set it up to have two masters: instruction bus and data bus.  Unfortunately, while the axi_lite_xbar has arbiters in it, reads and writes are separated.  When I have the instruction bus (master 0) doing lots of reads, and the data bus (master 1) try to do a write, it seems to be permanently stalled.  Is it possible to arbitrate reads and writes across all masters?

- Brendon

bchetwynd, 2018-10-17:
Since Github won't let you delete issues, I merely need to close it.  There is an issue with my peripheral... not the crossbar.

closed 2018-10-17 by bchetwynd

zarubaf, 2018-10-17:
Okay, still happy to help if you run into troubles.

bchetwynd, 2018-10-17:
Florian,

 

Much appreciated.  I will say that after exploring things a bit more, it seems like the arbiter should arbitrate between read and write requests.  Otherwise, I have can have concurrent requests going to a single component.  I need to dig into the AXI4-Lite spec some more,, but I am guessing that concurrent operating the read and write channels is probably not allowed….

 

Thank you,

Brendon

 

============================================================================

Brendon Chetwynd                                             Technical Staff

MIT Lincoln Laboratory                           

                                     Cyber Systems and Operations (Group 51)

 

244 Wood Street                                    Lexington, MA  02420-9185

                                                       781-981-8212 (office)

brendon.chetwynd@ll.mit.edu                            781-879-4635 (cell)

                                                       781-387-3030 (pager)

                                                       781-981-7548 (fax)

============================================================================