pulp-platform/axi #10: Add axi_cdc

accuminium, 2019-05-23:
Instantiate [`cdc_fifo_gray`](https://github.com/pulp-platform/common_cells/blob/master/src/cdc_fifo_gray.sv) for each of the AXI channels (expose `LOG_DEPTH` parameter) and add similar header/documentation.

accuminium, 2019-05-27:
@fabianschuiki Here is a first shot at the AXI CDC: https://github.com/pulp-platform/axi/blob/cdc/src/axi_cdc.sv. Simulation of the corresponding TB (https://github.com/pulp-platform/axi/blob/cdc/test/tb_axi_cdc.sv) looks good.

fabianschuiki, 2019-06-05:
I think this is exactly what we need! I especially like that the IP is more or less just a wrapper around multiple `cdc_fifo_gray` instances. We can ensure that these are functional and get the AXI CDC "for free" like this. Very cool!

accuminium, 2020-02-14:
This was implemented in #26.

closed 2020-02-14 by accuminium