pulp-platform/axi #21: Test driver compliance with AXI spec

chiraag, 2019-10-26:
The current test-driver in `axi_test.sv` waits for `axi.aw_ready` to be high in the `send_aw` task before trying the `send_w` task. I think this test-driver is not compliant with the AXI spec. AXI4 write response dependency sub-section in section A3.3 of the spec says that 

> the master must not wait for the slave to assert AWREADY or WREADY before asserting AWVALID or WVALID

> the slave can wait for AWVALID or WVALID, or both, before asserting AWREADY

Thus the test-bench can potentially fail an otherwise correct crossbar implementation. I might try to edit the test-driver to fix this and was wondering if you have any thoughts on a simple fix.

accuminium, 2019-10-27:
Hi Chiraag,

Thank you for your report.

Can you please link the behavior you are describing to code?  I understand that [the `send_aw` task of `axi_driver` waits for `aw_ready`](https://github.com/pulp-platform/axi/blob/d252bdf210cdf5e57b0cda3346739aedc7ac5f36/src/axi_test.sv#L375) before returning and therefore that waiting for `send_aw` to finish before calling `send_w` would violate the AXI specification.  However, `axi_driver` does not invoke these tasks itself but delegates that to users.

So the code in `axi_test.sv` is correct but it might be used incorrectly, right? If so, where?

fabianschuiki, 2019-10-27:
Indeed the "correct use" is currently left to the user. The `axi_test` driver only provides a simple interface to perform per-channel transactions. Maybe it would be a good idea to provide a `write` and `read` task which operate on an unsized queue `data_t data[$]` and send an entire sequence of data. In that task we could then also implement the dependency graph of AXI beats. What do you think, @accuminium -- is this a use case?

accuminium, 2019-10-27:
Yes, and there are developments towards higher-level AXI test drivers, e.g., [`rand_axi_master`](https://github.com/pulp-platform/axi/blob/190e80547d354857d4a7949ab472f647c5219074/src/axi_test.sv#L580). I think the driver you describe would fit nicely between the current, low-level `axi_driver` and the very high-level `rand_axi_master`. The instantiation chain could then be `rand_axi_master -> axi_master -> axi_driver` (`->` = instantiates), and users could instantiate one of these modules according to their required level of control. What do you think?

fabianschuiki, 2019-10-27:
This sounds excellent!

accuminium, 2019-10-27:
Perfect. Contributions welcome!

chiraag, 2019-10-27:
An example of incorrect use is in the `tb_axi_lite_xbar.sv` [here](https://github.com/pulp-platform/axi/blob/d252bdf210cdf5e57b0cda3346739aedc7ac5f36/test/tb_axi_lite_xbar.sv#L164-L168).

I was trying to use the test-bench to test a different xbar implementation when I noticed this issue.

accuminium, 2019-10-28:
I see, thanks for reporting.

I think this can be solved easily and have opened #22 to fix it. Can you test your XBAR implementation with that code?

accuminium, 2019-11-15:
Thanks for confirming the fix in #22.

closed 2019-11-15 by accuminium