pulp-platform/axi #39: AXI-lite crossbar status

ZipCPU, 2020-01-16:
I was running a formal verification check on the AXI-lite crossbar, and saw that it is no longer in the repository.  Can any one tell me the status of the core?  Should I expect it to be working?  or did it ever work?

Thanks!

zarubaf, 2020-01-16:
It should be working (in terms of AXI compliance). The main problem was that it never was a crossbar but only one master could be active at a time. We are working towards a reimplementation.

ZipCPU, 2020-01-16:
There were actually several issues associated with non-compliance.  [This is only one of them](https://zipcpu.com/img/tweets/axi-spec-registered.png).

zarubaf, 2020-01-16:
What was the exact path you could observe?

ZipCPU, 2020-01-16:
There were several.  For example rvalid coming from a slave returning combinatorially to the master.

zarubaf, 2020-01-16:
If you want you can cut those paths using an `src/axi_cut.sv`. I do not know what the rationale behind this sentence is though.

ZipCPU, 2020-01-16:
> I do not know what the rationale behind this sentence is though.

Are you asking about the rationale behind one of my sentences, or the sentence in the specification?

zarubaf, 2020-01-16:
The one in the spec. That would always induce a mandatory cycle delay in the crossbar, rather undesirable as a default imho.

ZipCPU, 2020-01-16:
It's actually worse than that.  It introduces not one but two mandatory cycle delays into the crossbar--one on the address channel and another on the return channel.

My best guess of the rational (since I didn't write the spec) is to keep any AXI logic off of the critical path in any time constrained system--helping you run at higher logic speeds.  Hence, if all the AXI logic outputs are registered, vs combinatorial, ... you get the idea.  The sad reality though is that it adds an additional level of complexity to any AXI design.

accuminium, 2020-02-11:
Thanks for your inputs regarding formal verification of our modules, we highly appreciate them!  We have just merged a new AXI4-Lite crossbar (#41), which actually *is* a fully-connected crossbar. Would you be interested in providing formal verification inputs on it?

ZipCPU, 2020-02-12:
Have you made the deliberate decision to allow _outputs_, such as the return values to the requesting masters, depend combinatorially upon the _inputs_ from the slaves?  I'm particularly discussing the R* channel, where the master can receive the data return on the same clock as the slave returns it.  I haven't checked any of the other channels for the same bug.

Other than that issue, I'm not seeing anything more.

Dan

accuminium, 2020-02-12:
> Have you made the deliberate decision to allow outputs, such as the return values to the requesting masters, depend combinatorially upon the inputs from the slaves?

Yes, one can opt-in to breaking all combinatorial paths by inserting `axi_lite_cut`s (as Florian pointed out).

> Other than that issue, I'm not seeing anything more.

Great to hear!  Is your formal verification setup for AXI-Lite available open-source?

ZipCPU, 2020-02-12:
The AXI-lite properties are all open source, yes.  However, I did build a wrapper around this module to instantiate those properties, and an SBY file that would bring in all of the various inputs.  I have not shared that wrapper with anyone.  We can discuss sharing that if you see yourself repeating this work.

The SystemVerilog extensions used by this AXI-lite crossbar are not supported by the open source version of SymbiYosys, but rather the commercial Symbiotic EDA Suite.

Dan

accuminium, 2020-02-12:
It would be great to have formal verification properties for the modules in this repository (in addition to the already existing, stimuli-based testbenches). Do you know if the input files for the commercial Symbiotic EDA suite can be shared under open-source licenses?

ZipCPU, 2020-02-12:
Sure, that's not a problem.  The [AXI-lite master properties are already open source and can be found here](https://github.com/ZipCPU/wb2axip/blob/master/bench/formal/faxil_master.v), next to their associated [AXI-lite slave properties](https://github.com/ZipCPU/wb2axip/blob/master/bench/formal/faxil_slave.v).

As for the other files, what would you recommend as the best place to put them?  They would be unique to this particular design.

accuminium, 2020-02-13:
How about creating a new directory `formal` in `test`? Would we add your `wb2axip` repository as a submodule (*) there to have access to the other properties, and add the files that are unique for this repository directly?

@zarubaf What do you think?

(*) As far as I can judge, the Apache 2.0 license of your repo should be compatible with the Solderpad 0.51 license (which is based on Apache 2.0) of our repo, but I am not a lawyer.

accuminium, 2020-11-12:
@ZipCPU What do you think?  Would you be willing to contribute your formal verification properties for the modules in this repository in a PR?

closed 2021-01-11 by accuminium