pulp-platform/axi #53: Naming Scheme for IDs and Wrappers

zarubaf, 2020-02-18:
Can we get a small contribution/style section on how we decided to name ports, interface wrapper and parameters of the modules? What do you think?

fabianschuiki, 2020-02-18:
That's an excellent idea. Should we create a `CONTRIBUTING.md` file for this, or just add a section to the readme? We could also add a link to coding style guidelines.

accuminium, 2020-02-18:
Good idea, thanks for the suggestion. How about adding a `CONTRIBUTING.md` that generally refers to the SystemVerilog Style Guide we use (by LowRISC) and specifically states that

- User-facing modules _must_ have SystemVerilog structs as AXI ports.
- User-facing modules _may_ come with a variant that has SystemVerilog interfaces as AXI ports. Such a variant module _must not_ implement any functionality except wiring its interfaces to the struct ports of the original module. The name of a variant _must_ be the name of the original module suffixed by `_intf`. Parameters of a variant must be formatted `ALL_CAPS`.

The rationale behind `ALL_CAPS` parameters for `_intf` modules is to provide a uniform "legacy" interface. I intend to change this for all `_intf` modules at once with the v1.0 release.

What do you think? Anything missing?

fabianschuiki, 2020-02-18:
@accuminium: This sounds like a solid plan!

zarubaf, 2020-02-18:
@accuminium That is perfect. Exactly what I was looking for. That can act as quick reference when implementing new modules! Thanks.

PR #54 (2020-02-18): Add Contribution Guidelines
This PR adds Contribution Guidelines, as discussed in #53.

Closes #53.
closed 2020-02-18 by accuminium