pulp-platform/axi #57: Data width converter violates AXI protocol on waiting for `aw_valid`

bluewww, 2020-02-18:
In `axi_dw_downsizer.sv`, the write channel does not proceed until the write address channel has finished its handshake (see [here](https://github.com/pulp-platform/axi/blob/1a1e41235d094562dd98af5a3d790ee41bb1f89c/src/axi_dw_downsizer.sv#L508)):
```sv
   // Can start a new request as soon as w_state_d is W_IDLE
    if (w_state_d == W_IDLE) begin
      // Reset channels
      w_req_d.aw = '0;

      if (slv_req_i.aw_valid && slv_req_i.aw.atop[5]) begin // ATOP with an R response
        inject_aw_into_ar_req = 1'b1                 ;
        slv_resp_o.aw_ready   = inject_aw_into_ar_gnt;
      end else begin // Regular AW
        slv_resp_o.aw_ready = 1'b1;
      end

      // New write request
      if (slv_req_i.aw_valid && slv_resp_o.aw_ready) begin
        // Default state
        w_state_d = W_PASSTHROUGH;

```
This is a violation of the AXI 4 protocol where a slave can wait for both `aw_valid`  and `w_valid` to be asserted.

fabianschuiki, 2020-02-18:
Good catch!

accuminium, 2020-02-19:
Thanks for reporting this.  Please note that this affects an unreleased module under development, but we will make sure this is addressed before releasing it.

suehtamacv, 2020-02-20:
The latest commit of the `axi_dwc_publish` branch, namely fa23b3a, proposes a (non-intentional?) work-around for this issue. 

This commit makes so that the DWC sends the AXI requests it doesn't support to an `axi_err_slv`. It does so via an `axi_demux`, which has a spill register at its write address channel. This spill register should solve the handshake issue discussed above.

This is more a fyi comment, and I will look into this issue soon.

suehtamacv, 2020-02-28:
5101ccb uses the spill registers instantiated by the AXI Demux at the AR/AW channels to solve the issue. The spill register absorbs the address write beat and sends `aw_ready` regardless of the status of the write channel.

accuminium, 2020-02-28:
Okay, thanks for resolving this. Can you please add a comment next to the spill parameters https://github.com/pulp-platform/axi/blob/5101ccbece7b08729441df1c4609f64d97e21a2a/src/axi_dw_downsizer.sv#L218-L219 stating that these registers (or only the AW?) are required for protocol compliance (i.e., to break the dependency of `w_valid` on `aw_ready`)? Without this comment, we might be tempted to change the parameters later when optimizing latency and/or throughput of this module.

suehtamacv, 2020-03-01:
Done.

closed 2020-03-01 by suehtamacv