pulp-platform/axi #60: DWC (upsizer) stops accepting non-modifiable ARs

accuminium, 2020-02-28:
This affects the DWCs under development in #43 (unreleased). The problem can be reproduced within an internal project.

suehtamacv, 2020-03-01:
Upon further analysis, this has been tracked to the `axi_atop_filter` not fully respecting the AXI handshake.

suehtamacv, 2020-03-01:
While not a bug in the data width converter, this behavior was triggered by the `axi_dw_upsizer` upsizing single beat transactions. This meant that, after further downsizing to the original bus width, this transaction could either still be a single burst transaction, or a burst of `len == 1`. These behave differently at the `axi_burst_splitter`, upstream `axready` depending or not on the downstream `axready`, which triggered the bug on the `axi_atop_filter`. 

https://github.com/pulp-platform/axi/blob/5b154041a4a29d49b029811f4498da990fe453b3/src/axi_burst_splitter.sv#L400-L425

664886b stops upsizing single-beat transactions and should trigger the bug regardless of the transaction `awcache` value.

suehtamacv, 2020-03-11:
Should this issue be closed, since the fix to the `axi_atop_filter` is now upstream?

accuminium, 2020-03-11:
Yes. Just to double-check, the upsizer no longer changes the `size` attribute of single-beat bursts, even if the burst is modifiable, right?

suehtamacv, 2020-03-11:
Indeed, the `size` field is not changed.

accuminium, 2020-03-11:
Perfect, thanks!

accuminium, 2020-03-11:
/close

closed 2020-03-11 by accuminium