pulp-platform/axi #63: DWC Slave Error on Burst Fixed with `len == 0`

WRoenninger, 2020-02-28:
This affects the DWCs under development in #43 (unreleased). The problem can be reproduced within an internal Ariane project.
As part of this I am using the DWC to connect APB peripherals as they have a data width of 32 instead of the 64 bit native to Ariane.
I have the issue that `Ariane` is issuing AXI transfers with `len == 0` as FIXED:
``` 
    // address
    // in case of a single request or wrapping transfer we can simply begin at the address, if we want to request a cache-line
    // with an incremental transfer we need to output the corresponding base address of the cache line
    assign axi_req_o.ar.burst  = (rd_blen_i == 0)      ? 2'b00 :
                                                         2'b01;
```
However currently the DWC responds to all `BURST_FIXED` with a slave error even though it should be possible to let this type of transactions through, as they behave the same as if the burst type would be `BURST_INCR`.
```
        if (w_req_d.aw.burst inside {BURST_WRAP, BURST_FIXED}) begin
          w_state_d              = W_PASSTHROUGH;
          w_req_d.aw_throw_error = 1'b1         ;
        end
```

suehtamacv, 2020-03-01:
While `FIXED` transactions with `len == 0` do have the same behavior as `INCR`, I am against adding this in the DWC. 

Downsizing `FIXED` transactions will require specific conversion logic in the DWC. For example, an incoming `FIXED` transaction wider than the outcoming bus needs to be converted into a sequence of `INCR` bursts. I would rather add the conversion logic for `FIXED` transactions as a whole, than adding an exception for this case, but I would like to hear @accuminium's opinion on that. As you said, there is an easy fix on Ariane's end, since `FIXED` and `INCR` bursts are equivalent.

accuminium, 2020-03-09:
I do not see the benefit of rejecting transfers we actually support, as is the case with `burst == Fixed` and `len == 0`. So I think we should keep stating that we do not support _Fixed_ bursts but only reject (i.e., return slave errors) them if they contain multiple beats.

suehtamacv, 2020-03-11:
a1931fe and dec4bbd create exceptions for single-beat transactions. @WRoenninger, can you check they fit your needs?

WRoenninger, 2020-03-12:
The exceptions seem to work in my simulations and the FPGA implementation of Ariane.

accuminium, 2020-03-13:
Thanks @suehtamacv for implementing and @WRoenninger for feedback!

closed 2020-03-13 by accuminium