pulp-platform/axi #68: Verilator compile issues

WRoenninger, 2020-03-09:
I am working on simulating the AXI4 code in Verilator and are running into a problem with the parameter types of the modules. The issue is the `parameter type resp_t = logic` definition in al modules. However renaming the definitions to `parameter type rsp_t = logic` seems to not trigger the issue. I think the issue could be that in `axi_pkg` the parameter `resp_t` is already defined and part of the response struct which is the parameter which has this problem. 
Should I do a fix for all modules, this will however break all other projects which are using the new AXI4 infrastructure currently. 

Related to Verilator issues: In #43 there is in file `src/axi_dw_upsizer.sv` line 321 a struct definition for `r_req_d` and `r_req_q` inside a generate. Verilator complains about multiple type definitions.
I was able to fix this by declaring a separate typedef outside the generate and defining the signals separate in the generate:

```systemverilog
typedef struct packed {
      ar_chan_t ar         ;
      logic ar_valid       ;
      logic ar_throw_error ;
      slv_r_chan_t r       ;
      logic r_valid        ;
      burst_len_t burst_len;
      size_t orig_ar_size  ;
} r_req_t;

// code

for (genvar ...
  r_req_t r_req_d, r_req_q;
end
```

accuminium, 2020-03-09:
### `resp_t` vs `axi_pkg::resp_t` conflict

I agree that using `resp_t` as parameter to hold the entire response struct is not ideal, since it could collide with `axi_pkg::resp_t` when the latter is imported into module namespace.  However, with namespaces guarding from exactly such conflicts, I am against changing this to `rsp_t` right now. Can you check in which modules exactly the conflict happens?

For instance, there are wildcard imports in `axi_cdc`, `axi_intf`, `axi_test`, `tb_axi_cdc`, and `tb_axi_atop_filter`.  Is one of those imports not bounded by a module scope?  I would prefer to fix this problem this way.

### Typedefs in `axi_dw_upsizer`

I agree with your solution. Please open a PR against the DWC branch.

WRoenninger, 2020-03-09:
I had the problem also for example in `axi_demux`. There no import is happening.

accuminium, 2020-03-09:
I see, but if one of the instances above is compiled before `axi_demux` and has a globally visible `import`, that could cause the problem, no?

WRoenninger, 2020-03-09:
This could be.

fabianschuiki, 2020-03-09:
Nice catch and fix! The compilation units in SV are really a pain.

accuminium, 2020-03-11:
This can be closed now that #70 is merged, right?

WRoenninger, 2020-03-12:
Yes #70 solved the compile issues.

PR #74 (2020-03-13): axi_intf: Improve Verilator compatibility
Closes #68.
closed 2020-03-13 by accuminium