pulp-platform/axi #72: Naming convention for interface ports

accuminium, 2020-03-11:
All our `interface`s have two `modports`: `Slave` and `Master`. This is aligned with the AXI terminology (and the only reason we are calling them like that). The ports themselves, however, are called `slv`/`mst` in 7/6 modules, `in` in 4/5 modules, and `src` in 1/1 module.  I think this is confusing and we should stick to a single convention. In alignment with the AXI standard and the majority of our modules, I propose `slv` and `mst`.

The situation is much better for request and response structs. We consistently use `slv`/`mst` with the exception of one module, `axi_cdc`.

Opinions?

accuminium, 2020-03-11:
/cc @zarubaf @fabianschuiki @WRoenninger

zarubaf, 2020-03-11:
That sounds good to me. 

Regarding the `cdc` module I am a bit inclined towards `src`/`dst` mainly because it denotes the source and destination clock domain. But I can certainly see it working with `slv` and `mst` as well, especially if the clocks and resets are named accordingly.

accuminium, 2020-03-11:
> Regarding the `cdc` module I am a bit inclined towards `src`/`dst` mainly because it denotes the source and destination clock domain.

I see and agree with that point.  Actually in one protocol converter, `axi_lite_to_apb`, we use `axi_lite_` and `apb_` as prefix, which makes perfect sense to me as well.  Should we harmonize this as well?

WRoenninger, 2020-03-11:
I also think we should stick to the AXI terminology and bring the naming of the interfaces in alignment to the struct version.
For the `cdc` I think we should also name it `mst` and `slv`, the reason is that not all channels follow instantiate same direction for the underlying `cdc_fifo` where it makes sense for the `src`/dst naming.

WRoenninger, 2020-03-11:
For the structs, should we take this opportunity to also redefine the `req`/`resp` to be something with the same length?

accuminium, 2021-01-20:
Superseded by #153; let us continue the discussion there.

closed 2021-01-20 by accuminium