pulp-platform/axi #77: `rand_id_queue_pkg` package missing?

zarubaf, 2020-03-18:
It seems that the `rand_id_queue_pkg` package is missing (used in https://github.com/pulp-platform/axi/blob/c803e7adac2e0df8a9e165603c3413cf12806e14/src/axi_test.sv#L1147). Is that intentional?

accuminium, 2020-03-18:
`rand_id_queue_pkg` is defined in https://github.com/pulp-platform/common_verification/blob/v0.1.1/src/rand_id_queue.sv, and this repository depends on `common_verification`: https://github.com/pulp-platform/axi/blob/ec6705502a493b7d672d137eac4ef0df0f27d008/Bender.yml#L11

Can you check whether your build environment satisfies this dependency?

zarubaf, 2020-03-18:
Thanks, that was it. Not sure why I thought it had to be in this repository.

closed 2020-03-18 by zarubaf