pulp-platform/axi #78: Questa Internal Error

zarubaf, 2020-03-18:
Unfortunately, with the newest Questasim Version (`Questa Sim-64 vsim 2020.1 Simulator 2020.01 Jan 28 2020`) this

https://github.com/pulp-platform/axi/blob/ec6705502a493b7d672d137eac4ef0df0f27d008/src/axi_xbar.sv#L217-L222

causes an internal error:

```
# ** Error:  src/axi/src/axi_xbar.sv(225): Questa has encountered an unexpected internal error: ../../src/vlog/vgentd.c(303). Please contact Questa support at http://supportnet.mentor.com/
```

Not sure how to solve this (and not suggesting to immediately react to it).

accuminium, 2020-03-18:
When you comment those lines out, it simulates fine?

zarubaf, 2020-03-18:
Affirm

accuminium, 2020-03-19:
Thanks. I am reluctant to add a version-specific `ifdef` into our code because it increases complexity with very narrow benefit. So please fix this locally for now.

Let's collect regressions in Questa 2020.1 in this thread.  I expect we find a pattern.  That would allow us to file a bug report to Mentor or implement an efficient workaround in our code.

accuminium, 2020-04-09:
> I am observing a major bug in an axi_mux instance in Questa 2020.1.
> All outgoing master request handshake signals are X -

Thanks for reporting, this sounds very serious!

>  but I can observe their direct source (spill registers) as being non-X.
> Interestingly, it happens only in one of two instances in the design; it might have something to do with the spill registers which only that instance has, I will investigate further today.
> Edit: disabling (and also manually coding out) spill registers made no difference.

Did you find out more about the failure mechanism?

da-gazzi, 2020-04-14:
> Did you find out more about the failure mechanism?
Not yet - I'm currently working on reproducing it in a more isolated setup. I'll report back if I find anything!

zarubaf, 2020-04-14:
At the integration meeting today we were running most of the AXI infrastructure on QuestaSim 2020.1 with no problems. I am starting to wonder whether there might be a problem with our internal installation.

PR #140 (2021-01-12): CI: Test multiple recent vsim versions
CI now includes Questa 2020.1. This PR therefore closes #78, as all tests pass also with Questa 2020.1.
closed 2021-01-12 by accuminium