pulp-platform/axi #90: Signals AR_READY / AW_READY on Master side

SebastianZa, 2020-04-08:
Hi all,

currently I do a simulation on your AXI XBAR with 2 masters and 2 slaves.

There I'm facing a problem with the signals A[R|W]_READY on master interfaces. At the start of the simulation they are '1' and they remain '1' during the simulation. The two slaves I wrote set their READY signals to '0' at the start of the simulation and behave correctly. The ready signals for the data and the WriteResponse channel behave correctly on master and slave side.

I ran the original TB tb_axi_xbar and there I saw the same behavior. Could you please provide help here?

I tried to attach 3 screenshots that describe the situation in detail. The waveforms are generated with the original testbench. Unfortunately I was not able to upload them. I will try later on.

Kind regards
Sebastian

SebastianZa, 2020-04-08:
Now it worked with the upload. :)

[image]
[image]
[image]

WRoenninger, 2020-04-09:
Hi Sebastian, 

What you are seeing is allowed as per AXI specification (Chapter A3.2.1, Figure A3-3). The `master` interface of the testbench connects the master (source) port of a `rand_axi_master` to a slave (destination) port of the `axi_xbar`. The `axi_xbar` is thereof allowed to assert the A[W|R]_READY signals as long as it is capable of handling new beats on the corresponding channel.

The `axi_xbar` is written in a way that makes it possible to have multiple A[W|R] transfers in flight at the same time, hence the constant READY as the internal buffers/FIFOs are not saturated after initiation of a single A[W|R] transfer.

I hope this helps. :)

Best,
Wolfgang

SebastianZa, 2020-04-09:
Hi Wolfgang,

thank you very much for your fast reply! :)

This part of the specification is understood. What I don't understand is that the data on the appropriate channel is transmitted from master to slave before the address which belongs to the specific data (sorted by the ID field).

I can see in the testbench that from one point in time data to the channel is transmitted but not the address. I would expect that the data on this channel is hold back in a FIFO as well? At that Point when data is first transmitted to the slave interface without changing the address I get an exception that a counter underflow occurred possible due to an ID error.

I hope the explanation is not to confusing. If you don't mind I will provide a detailed explanation with two more screenshots next Wednesday after Easter (currently I don't have access to my simulation workstation).

Kind regards
Sebastian

WRoenninger, 2020-04-09:
Hi Sebastian,

There are two possible explanations I can think of right now. 
However it would be nice, if you could provide your `axi_pkg::xbar_cfg_t` (xbar configuration) and `axi_pkg::xbar_rule_[64|32]_t` array (address map) declarations, as these have implications on the instantiation and behavior of the xbar.

If I understood correctly this issue comes up with your slave module? 
* We have support for the atomic transactions from AXI5 (ATOP signal, Chapter E1.1). In the pictures I see that there is a change in the ATOP field of the AW transaction. These control signals from the atomics are transferred only the AW channel, however generate responses on the R channel if the highest bit (5) is set. The crossbar then expects a read response from a connected slave even if no AR has been sent. This could be the issue with the counter underflow. If your slave module is not capable of handling atomics, then you can do two things:
    * Instantiate an `axi_atop_filter` between the xbar and your module, this filters atomic transactions out and replies correctly with an error.
    * Set the parameter `AXI_ATOPS` of the `rand_axi_master` to `1'b0`, then no atomic transactions get generated.

* The crossbar sees a transaction as started as soon as the AW_VALID signal on a slave port is high. Then we know the contents of the AW vector and how to route the write data. From then on it connects the W channel to the appropriate slave until the beat with the W_LAST flag is transferred. This happens regardless if the AW vector is transferred away from the crossbar or not, as the information for the write data routing is already there. The `rand_axi_slave` module from `axi_test` will receive W beats regardless if an AW is transferred or not. It will however sends the B response only if all W beats to an AW beat are received. 

Best,
Wolfgang

SebastianZa, 2020-04-15:
Hi Wolfgang,

thanks for your quick and great help! I had a few mistakes in my signals (len / burst) and this was the reason that the intercon stalled at some time.

I keep working with your AXI implementation. Next step is connect my own master (not only simulation model but real master) and two slaves to the intercon.

Great work!

Kind regards
Sebastian

closed 2020-04-15 by SebastianZa