pulp-platform/axi #105: Consider checking rtl with `vlog -lint -pedanticerrors`

bluewww, 2020-05-04:
Some code does not compile with `-pendaticerrors` (strict langauge checks) for example
```
git clone git@github.com:pulp-platform/common_verification.git
vlog -lint -pedanticerrors common_verification/src/rand_id_queue.sv
vlog -lint -pedanticerrors src/axi_pkg.sv
vlog -lint -pedanticerrors src/axi_test.sv
```
results in
```
** Error (suppressible): src/axi_test.sv(72): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.
** Error (suppressible): src/axi_test.sv(356): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.
** Error (suppressible): src/axi_test.sv(965): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.
** Error (suppressible): src/axi_test.sv(1207): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.
** Error (suppressible): src/axi_test.sv(1357): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.
** Error (suppressible): src/axi_test.sv(1514): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.
** Error (suppressible): src/axi_test.sv(1683): (vlog-13262) A virtual interface element is not allowed in a sensitivity list.
```

I believe it is useful to limit ourselves to the sv standard, since it will improve compatibility between different vendor tools.

accuminium, 2020-05-05:
Agreed, thanks for reporting.

accuminium, 2021-01-12:
#141 will introduce linting and pedantic error checking for all code in `src/` -- except the `axi_test` package, where you reported problems.

I quickly tried to resolve those problems (patches at the bottom; feel free to continue investigating):

The reports are probably correct; the SV LRM (Section 25.9 in 1800-2012) says "they [=the components of virtual interfaces] cannot be used in continuous assignments or sensitivity lists. In order for a net to be driven via a virtual interface, the interface itself must provide a procedural means to do so. This can be accomplished either via a `clocking` block or by including a driver that is updated by a continuous assignment from a variable within the interface."

So I added clocking blocks to both of our `_DV` interfaces (first two patches) and modified the sensitivity lists that raise errors to be sensitive on the clocking block instead of the signal component (as also described [in this example](https://verificationacademy.com/forums/uvm/virtual-interface-element-not-allowed-sensitivity-list#answer-88922)) (third patch). However, QuestaSim (I tested 2020.1) still raises the same errors. In fact, an [answer to the example](https://verificationacademy.com/forums/uvm/virtual-interface-element-not-allowed-sensitivity-list#answer-89711) states:
> l get this error in questa only if the ‘pedanticerrors’ flag is set.  [The example] code will also flag this if compiled with this flag. However, I still think it’s a bug in Questa [...]

So this seems to be yet another poorly supported feature of SV, and I suggest we wait a couple of decades before trying to adopt it again.

```diff
From f67af95de2bc4af46425ce6375997461d732b963 Mon Sep 17 00:00:00 2001
From: Andreas Kurth <akurth@iis.ee.ethz.ch>
Date: Tue, 12 Jan 2021 15:55:31 +0100
Subject: [PATCH 1/3] [wip] AXI_LITE_DV: Define clocking blocks

TODO: Are the defined default skews what we want?
---
 src/axi_intf.sv | 14 ++++++++++++++
 1 file changed, 14 insertions(+)

diff --git a/src/axi_intf.sv b/src/axi_intf.sv
index 900a33c210..8314b2427b 100644
--- a/src/axi_intf.sv
+++ b/src/axi_intf.sv
@@ -443,4 +443,18 @@ interface AXI_LITE_DV #(
     output r_data, r_resp, r_valid, input r_ready
   );
 
+  clocking Master_clocking @(posedge clk_i);
+    default input #1step output #0;
+    input  aw_ready, w_ready, b_resp, b_valid, ar_ready, r_data, r_resp, r_valid;
+    output aw_addr, aw_prot, aw_valid,  w_data, w_strb, w_valid, b_ready,  ar_addr, ar_prot,
+           ar_valid, r_ready;
+  endclocking
+
+  clocking Slave_clocking @(posedge clk_i);
+    default input #1step output #0;
+    input  aw_addr, aw_prot, aw_valid,  w_data, w_strb, w_valid, b_ready,  ar_addr, ar_prot,
+           ar_valid, r_ready;
+    output aw_ready, w_ready, b_resp, b_valid, ar_ready, r_data, r_resp, r_valid;
+  endclocking
+
 endinterface
-- 
2.30.0
```

```diff
From d5f53424a01539085ed9f8260b8bc2cb336c645a Mon Sep 17 00:00:00 2001
From: Andreas Kurth <akurth@iis.ee.ethz.ch>
Date: Tue, 12 Jan 2021 16:00:54 +0100
Subject: [PATCH 2/3] [wip] AXI_BUS_DV: Define clocking blocks

TODO: Are the defined default skews what we want?
---
 src/axi_intf.sv | 29 +++++++++++++++++++++++++++++
 1 file changed, 29 insertions(+)

diff --git a/src/axi_intf.sv b/src/axi_intf.sv
index 8314b2427b..794a067e49 100644
--- a/src/axi_intf.sv
+++ b/src/axi_intf.sv
@@ -192,6 +192,35 @@ interface AXI_BUS_DV #(
           r_id, r_data, r_resp, r_last, r_user, r_valid, r_ready
   );
 
+  clocking Master_clocking @(posedge clk_i);
+    default input #1step output #0;
[... 96 lines cut ...]
+      repeat (cycles) @(this.drv.axi.Master_clocking);
     endtask
 
     task automatic send_ars(input int unsigned n_reads);
@@ -1538,7 +1540,7 @@ package axi_test;
         cycles <= max;
       };
       assert (rand_success) else $error("Failed to randomize wait cycles!");
-      repeat (cycles) @(posedge this.drv.axi.clk_i);
+      repeat (cycles) @(this.drv.axi.Slave_clocking);
     endtask
 
     task automatic recv_ars();
@@ -1709,7 +1711,7 @@ package axi_test;
 
     /// End this cycle
     protected task automatic cycle_end;
-      @(posedge this.axi.clk_i);
+      @(this.axi.Monitor_clocking);
     endtask
 
     /// Handle update of the golden model with W beats.
-- 
2.30.0
```

PR #141 (2021-01-12): Compile with linting and pedantic error checking
Closes #105.
closed 2021-01-12 by accuminium