pulp-platform/axi #117: Vivado simulator 2020.1 segmentation fault

jeras, 2020-06-30:
I tried to simulate `axi_lite_xbar` with Xilinx VIP using the Vivado simulator.
during the `xelab` stage I got a segmentation fault, so I started to comment out sections of code.

I found out commenting out the instance of `axi_lite_to_axi` was kind of enough to get the `xelab` to not crash.
https://github.com/pulp-platform/axi/blob/master/src/axi_lite_xbar.sv#L165-L180

The converter itself does not have much code, and commenting it out did not help. So I started commenting out ports (both in the module and at the instance).
https://github.com/pulp-platform/axi/blob/master/src/axi_lite_to_axi.sv#L26-L27
Now I got an error I can not explain yet:
```
Starting static elaboration
Pass Through NonSizing Optimizer
ERROR: [VRFC 10-3814] value of 'req_lite_t' depends on itself [.../axi/src/axi_lite_to_axi.sv:19]
ERROR: [VRFC 10-3814] value of 'resp_lite_t' depends on itself [.../axi/src/axi_lite_to_axi.sv:20]
ERROR: [XSIM 43-3322] Static elaboration of top level Verilog design unit(s) in library work failed.
```
I tried to remove the default value of the parameter but I only got 2 extra warnings for the missing default value.

**WORKAROUND:** I renamed the two parameters by appending 1 at the, end and now `xelab` seems to get through.

I did not try to debug this further but I can speculate.
A. There is something globally defined with the same name as the parameter.
B. Vivado has some issue properly separating namespaces.

I was also compiling some VHDL code in the same project (actually just a barebone shell script).
I commented out the `xvhdl` command to avoid the VHDL namespace but it did affect the segmentation fault.

I do not have a relationship with Xilinx support right now, would you be able to file a bug report?
I also do not have the time to prepare a small example right now.

accuminium, 2021-01-11:
Hi @jeras, thanks for reporting this issue. Could you share the TCL commands that you used to run Vivado simulator?

jeras, 2021-01-11:
Sorry, I do not think I will have the time to remove proprietary code to be able to provide you with a clean example.
I just created this issue, so that anybody else encountering it would be able to find at least one reference online, and maybe use the same workaround.

accuminium, 2021-01-12:
I see, thanks for the additional information.

As this affects neither Vivado synthesis nor other RTL simulators, I suspect it is a bug specific to the Vivado simulator.  We do not have the resources to track that down without a minimal failing example.

If someone really needs to use Vivado simulator 2020.1, please use the workaround indicated above.  Otherwise, please use another simulator.

I am closing this issue. Please feel free to re-open if there is evidence that this is a larger problem or someone has a minimal failing example that can be shared here and with Xilinx support.

closed 2021-01-12 by accuminium