pulp-platform/axi #120: there is no  "rand_id_queue_pkg"

CaseyZhu, 2020-07-24:
rand_id_queue_pkg::rand_id_queue"
  Target for scope resolution operator does not exist

bluewww, 2020-07-24:
It's [here](https://github.com/pulp-platform/common_verification), though I don't know why it is not listed as dependency.

accuminium, 2020-09-10:
This repository lists [`common_verification@v0.1.1` as dependency](https://github.com/pulp-platform/axi/blob/master/Bender.yml#L12), and [`rand_id_queue_pkg::rand_id_queue` is part of `common_verification@v0.1.1`](https://github.com/pulp-platform/common_verification/blob/v0.1.1/src/rand_id_queue.sv#L15).

I thus assume this is a problem with local dependency resolution. Please reopen the issue if there is a problem in this repository.

closed 2020-09-10 by accuminium

accuminium, 2020-09-11:
The `common_verification` dependency [was missing in `ips_list.yml`](https://github.com/pulp-platform/axi/blob/cb896dbf7693d90f5641167e3dfb2a6dc59828fc/ips_list.yml), so I assume this issue described that problem. This has now been fixed and will be part of the next release.