pulp-platform/axi #121: `axi_demux`: Slave port stalling if the master issues writes depending on reads.

WRoenninger, 2020-08-07:
There can be instances where the slave port on `axi_demux` deadlocks if the connected master module generates `AW` transactions depending on `AR` transactions (as can happen when #110 is connected to a `axi_xabr`). 

In addition the master has to be capable to launch more transactions than the `xbar` can forward (depending on `MaxMstTrans`).
Then the instantiated id counters of the `AR` in `axi_demux` can stall the `AW` path due to handling of the `ATOP` inject. This then causes the `AW`s to stall and not letting `W` through. This leads to stalling of the `R` channel, which causes the `AR` in the id counters not being retired, leading to a deadlock. 

This can at the moment be prevented by setting `MaxMstTrans` to a value such that the id_counters never start stalling.

accuminium, 2021-01-20:
So setting `MaxMstTrans` (defined as "Each slave port can have at most this many transactions in flight.") to at least the value of the number of ARs issued by that master plus one (for the AW) solves the issue?

WRoenninger, 2021-01-20:
When I remember correctly, yes setting the value high enought should not trigger the issue. 

Found it whilst having a discussion with @thommythomaso about #110. We where not sure how to resolve this natively. The protocol explicitly states that a port should not have a dependency between the channels, which a construct like #110 breaks by design on its master port.

accuminium, 2021-01-20:
Right, so this not a problem of `axi_demux` but a limitation of the protocol. It is nonetheless good to know that `axi_demux` can handle even this case provided `MaxMstTrans` is set to a sufficiently high value.

closed 2021-01-20 by accuminium