pulp-platform/axi #122: Problem with rand_load in tb_axi_lite_regs

dawidzim, 2020-09-07:
Hey!
In specific seed they value of `rand_load` in tests `tb_axi_lite_regs` causes assertion fail:
```
KERNEL: 1500000 Lite Master> Recieved read response from ADDR: 00000000 DATA: de01beef RESP: 0
KERNEL: Fatal Error: /home/tester/gitlab/axi/test/tb_axi_lite_regs.sv (124): Data is unexpected, should be axi_data_t'(64'hDEADBEEFDEADBEEF).
```
If we change seed, test works fine.

accuminium, 2020-09-10:
Hi @dawidzim, thanks for reporting this issue.

Can you please provide one seed value that causes the test to fail and one seed value that causes the test to pass, so we can reproduce the issue?

dawidzim, 2020-09-11:
Hey @accuminium !
Unfortunate each simulator have other value for specific seed, so I can't provide you this.
But you can try do this:
```
        axi_lite_regs)
            for PRIV in 0 1; do
                for SECU in 0 1; do
                    for SEED in {0..100}; do
                        call_vsim tb_axi_lite_regs -sv_seed $SEED -GPrivProtOnly=$PRIV -GSecuProtOnly=$SECU
                    done
                done
            done
            ;;
```
Maybe you will be able to catch this problem

accuminium, 2021-01-11:
Indeed, `-sv_seed 10` triggers the problem. Thanks @dawidzim!

This appears to be a limitation of the testbench. Between write and successive read, the TB loads one of the bytes in the register with a different value, but it does not update the expected bytes to reflect the change.

A short-term workaround is to disable the random loading in the `toggle_load` task of the TB, but that reduces the functional coverage.

@WRoenninger: Do you have a suggestion on how the TB could be extended to account for this?

WRoenninger, 2021-01-12:
Made a pull request in #139 which should fix the issue.

Indeed the issue is exactly how @accuminium described.

My solution is to remove the failing hardcoded data assertion associated to the read. This can be done safely as the data from every read is asserted in `proc_check_read_data` against the dynamic data model anyway. So in this case the data that triggered the assertion was actually checked against two differing models.

PR #139 (2021-01-12): Fix issue 122
This fixes #122.

The tb performs a known write followed by a read at the begining of simulation to a writeable address. The hardcoded data read assertion can be safely removed as the read data is checked by `proc_check_read_data` at the same time against the dynamic model.

Also added some more parameters in the vsim run script for the tb.
closed 2021-01-12 by accuminium