pulp-platform/axi #128: VCS compilation issue for AXI lite modules

saadkhalil283, 2020-11-12:
Please see the following error message shown by vcs

Error-[SV-UIP] Unconnected interface port
axi_lite_demux.sv, 421
"slv"
  The port 'slv' of top-level module 'axi_lite_demux_intf' whose type is 
  interface 'AXI_LITE' is left unconnected. It is illegal to leave the 
  interface ports unconnected.
  Please make sure that all the interface ports are connected.

accuminium, 2020-11-12:
> The port 'slv' of **top-level module** 'axi_lite_demux_intf' whose type is interface 'AXI_LITE' is left unconnected.

Apparently, VCS does not support SystemVerilog interfaces in top-level modules (which would be a restriction shared with many other tools).  However, `axi_lite_demux` is not designed to be a top-level module.  Are you sure that you want it to be the top-level module?  If not, you should change the top-level module in VCS.

saadkhalil283, 2020-11-17:
Yes, _axi_lite_demux_ is not the top module. _tb_axi_lite_xbar_ is the top module. The problem is while compiling through the hierarchy when VCS compiles _axi_lite_demux_intf_, the module contains an interface instance of _AXI_LITE_ for both its **Master** and **Slave** modports. That's where it raises this error. Same error is also raised in _axi_lite_mux_intf_ and _axi_to_axi_lite_intf_.

accuminium, 2020-11-17:
I tried to reproduce the error you reported but could not.  I was able to simulate `tb_axi_lite_xbar` in VCS 2020.03. Here is what I did:
1.  I checked out the newly-created [`vcs-improve-compatibility` branch](https://github.com/pulp-platform/axi/tree/vcs-improve-compatibility) (more on that below).
2.  I created a `build` directory and cd'ed into it.
3.  I used [Bender 0.21.0](https://github.com/fabianschuiki/bender/tree/v0.21.0) to generate a VCS compile script:
    ```sh
    bender script vcs -t test --vlog-arg='-assert svaext' --vlog-arg='-timescale=1ps/1ps' > compile_vcs.sh
    ```
4.  I made the `compile_vcs.sh` script executable and ran it.
5.  I compiled the `simv` binary for `tb_axi_lite_xbar` with
    ```sh
    vcs -full64 tb_axi_lite_xbar -debug_access+all -notice
    ```
6.  I ran the `simv` binary and saw the TB successfully run to completion.

The [`vcs-improve-compatibility` branch contains 7 commits on top of the current master branch (v0.24.1)](https://github.com/pulp-platform/axi/compare/v0.24.1...vcs-improve-compatibility). All of them are required: VCS will not run `tb_axi_lite_xbar` on v0.24.1 due to multiple issues, but the error you reported is not one of them.
- The first five commits are quite generic changes to work around SystemVerilog limitations in current VCS.  I would have to test them in other simulators, but in general they could be mainlined.
- The last two commits (https://github.com/pulp-platform/axi/commit/4c360a9efd5e02769642681466a8371e8e8e9249 and https://github.com/pulp-platform/axi/commit/74af938e273a4bfce7c2c9b926e636419aeb4239), however, are required to work around a more serious problem/bug of current VCS: It does not correctly pass the value of a parameter to instantiated classes, at least in how we instantiate them in this case. Those commits are not going to be mainlined as they are, because they *mask* a problem with VCS (the default values are correct for `tb_axi_lite_xbar`, but they may not be for others), rather than *solving* it.

Please try the steps above.  If they do not work for you, please post your error message.  If they do work, I would be very happy about a PR that really solves (some of) the problems with VCS.

saadkhalil283, 2020-11-20:
> I tried to reproduce the error you reported but could not. I was able to simulate `tb_axi_lite_xbar` in VCS 2020.03. Here is what I did:
> 
>     1. I checked out the newly-created [`vcs-improve-compatibility` branch](https://github.com/pulp-platform/axi/tree/vcs-improve-compatibility) (more on that below).
> 
>     2. I created a `build` directory and cd'ed into it.
> 
>     3. I used [Bender 0.21.0](https://github.com/fabianschuiki/bender/tree/v0.21.0) to generate a VCS compile script:
>        ```shell
>        bender script vcs -t test --vlog-arg='-assert svaext' --vlog-arg='-timescale=1ps/1ps' > compile_vcs.sh
>        ```
> 
>     4. I made the `compile_vcs.sh` script executable and ran it.
> 
>     5. I compiled the `simv` binary for `tb_axi_lite_xbar` with
>        ```shell
>        vcs -full64 tb_axi_lite_xbar -debug_access+all -notice
>        ```
> 
>     6. I ran the `simv` binary and saw the TB successfully run to completion.
> 
> 
> The [`vcs-improve-compatibility` branch contains 7 commits on top of the current master branch (v0.24.1)](https://github.com/pulp-platform/axi/compare/v0.24.1...vcs-improve-compatibility). All of them are required: VCS will not run `tb_axi_lite_xbar` on v0.24.1 due to multiple issues, but the error you reported is not one of them.
> 
>     * The first five commits are quite generic changes to work around SystemVerilog limitations in current VCS.  I would have to test them in other simulators, but in general they could be mainlined.
> 
>     * The last two commits ([4c360a9](https://github.com/pulp-platform/axi/commit/4c360a9efd5e02769642681466a8371e8e8e9249) and [74af938](https://github.com/pulp-platform/axi/commit/74af938e273a4bfce7c2c9b926e636419aeb4239)), however, are required to work around a more serious problem/bug of current VCS: It does not correctly pass the value of a parameter to instantiated classes, at least [when defining and instantiating classes like this](https://github.com/pulp-platform/axi/blob/v0.24.1/test/tb_axi_lite_xbar.sv#L75-L94). Those commits are not going to be mainlined as they are, because they _mask_ a problem with VCS (the default values are correct for `tb_axi_lite_xbar`, but they may not be for others), rather than _solving_ it.
> 
> 
> Please try the steps above. If they do not work for you, please post your error message. If they do work, I would be very happy about a PR that really solves (some of) the problems with VCS.

Thanks a lot Sir. I followed the steps and was able to simulate the code in VCS 2018.03. I just had to add a couple of flags with the VCS command that you mentioned.

accuminium, 2020-11-20:
Perfect, thanks for your feedback.  I think we can conclude two aspects from this:

- Our **synthesizable code** has a few pieces that VCS does not understand (even though they are standard-compliant and synthesizers and other simulators understand them).  Those pieces can and will be fixed to improve compatibility with VCS, provided they do not break semantics for other tools, on a per-case basis.  I will create a PR to mainline the fixes developed in this issue.
- Our **test-only code** (i.e., TBs and the `axi_test` package) uses (probably only a few) features that are problematic for VCS (unsupported and/or buggy). There might be workarounds for them, such as for the problems discussed here. However, the workarounds found so far are TB-specific and cannot be applied to the entire repository. VCS experts are encouraged to contribute fixes.

closed 2020-11-20 by accuminium