pulp-platform/axi #129: Build and deploy documentation is failing.

WRoenninger, 2020-12-07:
When the CI runs, the build and deploy documentation fails with following output in `Determine documentation target folder`:

```bash
Run if echo $GITHUB_REF | grep -qE '^refs/(heads|tags)'; then
Error: Unable to process command '::set-env name=DOC_TARGET::axi_to_mem' successfully.
Error: The `set-env` command is disabled. Please upgrade to using Environment Files or opt into unsecure command execution by setting the `ACTIONS_ALLOW_UNSECURE_COMMANDS` environment variable to `true`. For more information see: https://github.blog/changelog/2020-10-01-github-actions-deprecating-set-env-and-add-path-commands/
```

WRoenninger, 2020-12-07:
On that note, it would be nice to add the link to the automatic generated documentation to the top README.

accuminium, 2020-12-18:
Indeed, thanks for reporting @WRoenninger!

PR #132 (2020-12-18): Fix auto-generated documentation and mention it in ReadMe
Closes #129.
commit c42eff6f (2020-12-18): workflows/doc: Replace deprecated `add-path`

See [1] and [2].  Fixes #129.

[1]: https://github.blog/changelog/2020-10-01-github-actions-deprecating-set-env-and-add-path-commands
[2]: https://github.com/actions/toolkit/security/advisories/GHSA-mfwh-5m23-j46w
closed 2020-12-18 by accuminium