pulp-platform/axi #130: Separating master and slave of AXI-lite

mirzasaadkhalil, 2020-12-10:
Hi,
I am working on AXI-complaint DMA. I have to use the master and slave of AXI-lite separately. I am a bit confused which modules will constitute the `master` and `slave`. If i am thinking in the wrong direction kindly correct me. Thanks a lot.
Best regards.

WRoenninger, 2020-12-14:
Hi,
`Master` and `Slave` are referring as per AXI4 specification to the port characteristics of a module. `Master` ports are issuing transactions towards `Slave` ports. I.e. a `master` port writes data to a `slave` port or reads data from it. A module can have multiple `master` and/or `slave`ports.

In case for a DMA it would usually have at least one AXI `master` port which will read data from an address on a `slave` port  inside the system and write it to another address using the same `master` port. The DMA module could for example also have an AXI `slave` port for configuration from a `master` port on a CPU.

The modules here on the master branch currently consist mainly of interconnect modules. `Master` ports have `mst` and `slave` ports have `slv` in their module port declarations. They are not directly intended to use inside a DMA implementation but to connect the ports of different modules featuring AXI ports inside a system (e.g. over an `axi_xbar` or `axi_lite_xbar`).

There is currently also a pull request #110 which is implementing a backend for an AXI compliant DMA. This backend is implementing the `master` port which will perform the data movements from one address to another inside a system.

Best regards.

accuminium, 2020-12-18:
Well explained, thanks @WRoenninger!

closed 2020-12-18 by accuminium