pulp-platform/axi #133: illegal port connection

haitianzhishang, 2020-12-20:
[image]

** Error (suppressible): (vsim-3053) /home/fyp/pulpissimo/sim/../ips/axi/axi/src/axi_xbar.sv(162): Illegal output or inout port connection for port 'slv_resp_o'.
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/i_soc_interconnect_wrap/i_soc_interconnect/i_axi_xbar/i_xbar/gen_slv_port_demux[6]/i_axi_err_slv File: /home/fyp/pulpissimo/sim/../ips/axi/axi/src/axi_err_slv.sv

This error comes when I'm running the PULP-rt-example.
Thank you

accuminium, 2020-12-21:
Thanks for reporting this issue, @haitianzhishang.  Am I correct to assume that you are using the latest version (`v6.0.0`) of `pulpissimo`?

At `v6.0.0`, [`pulpissimo` instantiates `pulp_soc` at `v2.0.0`](https://github.com/pulp-platform/pulpissimo/blob/v6.0.0/ips_list.yml#L21-L24), which in turn [instantiates `axi_xbar_intf` in `soc_interconnect`](https://github.com/pulp-platform/pulp_soc/blob/v2.0.0/rtl/pulp_soc/soc_interconnect.sv#L303-L316). This would match the path in the error message you posted. The [used version of AXI should be `v0.24.1`](https://github.com/pulp-platform/pulp_soc/blob/v2.0.0/ips_list.yml#L50-L52), which is the current version.

@bluewww and @meggiman as maintainers of PULP SoC and PULPissimo, have you ever seen this problem before?

bluewww, 2020-12-21:
I have not seen this problem before. Note that @haitianzhishang uses `vsim-10.6b` so it can also be a version specific problem.

haitianzhishang, 2020-12-21:
I have checked it. The Pulpissimo version I use is v6.0.0 and the version of AXI is also v0.24.1  @accuminium 
My vsim version is 10.6c exactly @bluewww 

And another error occurs when I run the accelerators example. If there is something run when I configure the environment.
[image]

Thanks a lot

bluewww, 2020-12-21:
For you second problem you need to pulp-sdk to run this example afaik.

haitianzhishang, 2020-12-22:
Yes, the second problem has been fixed. Thank you@bluewww

jsailer, 2021-01-08:
I am also working with pulpissimo v6.0.0 and get the exact same problem (`Illegal output or inout port connection for port 'slv_resp_o'.`).
And I am using Questa Sim-6 (Version 10.6c_1 linux_x86_64 Sep 15 2017)

accuminium, 2021-01-11:
It looks like @jsailer found a workaround! @haitianzhishang, can you check if https://github.com/pulp-platform/axi/commit/1ec8f0039f3e79fc92a97a5c744507e04406179a solves the problem for you too?

PR #138 (2021-01-12): axi_xbar: Improve compatibility with vsim 10.6c
This is a workaround for a problem with vsim 10.6c, see issue #133.
That version of vsim reports an "illegal output or inout port connection
for port 'slv_resp_o'" error (`vsim-3053`).  More recent versions of
vsim do not have this problem.

This workaround preserves the original functionality.

Thanks @jsailer for implementing this in https://github.com/jsailer/axi/commit/877364c130ed14dce4016410ea004008a118e7ca.

Fixes #133.
closed 2021-01-12 by accuminium