pulp-platform/axi #150: axi_burst_splitter: Index out of bound warning.

WRoenninger, 2021-01-19:
When running `scripts/run_vsim.sh tb_axt_to_axi_lite`, there are excessive warings caused by `axi_burst_splitter` line `545`, `551` and `559`.
Not sure if ci picks up on this, as the run script runs through regardless.

Example warnings:
```bash
# ** Warning: (vsim-8233) Index xxxx into array dimension [9:0] is out of bounds.
#    Time: 75621 ns  Iteration: 4  Process: /tb_axi_to_axi_lite/i_dut/i_axi_to_axi_lite/i_axi_burst_splitter/i_axi_burst_splitter_ar_chan/i_axi_burst_splitter_counters/#ALWAYS#549 File: /scratch/msc19f10/axi/src/axi_burst_splitter.sv Line: 551
# ** Warning: (vsim-8233) Index xxxx into array dimension [9:0] is out of bounds.
#    Time: 75621 ns  Iteration: 4  Process: /tb_axi_to_axi_lite/i_dut/i_axi_to_axi_lite/i_axi_burst_splitter/i_axi_burst_splitter_ar_chan/i_axi_burst_splitter_counters/#ALWAYS#557 File: /scratch/msc19f10/axi/src/axi_burst_splitter.sv Line: 559
# 75633 axi_lite_rand_slave> Recv  W with DATA: f92db34a SRTB: 0
# 75635 axi_lite_rand_slave> Recv AR with ADDR: 17599778 PROT: 000
# ** Warning: (vsim-8233) Index xxxx into array dimension [9:0] is out of bounds.
#    Time: 75635 ns  Iteration: 4  Process: /tb_axi_to_axi_lite/i_dut/i_axi_to_axi_lite/i_axi_burst_splitter/i_axi_burst_splitter_ar_chan/i_axi_burst_splitter_counters/#ALWAYS#557 File: /scratch/msc19f10/axi/src/axi_burst_splitter.sv Line: 559
# ** Warning: (vsim-8233) Index xxxx into array dimension [9:0] is out of bounds.
#    Time: 75637 ns  Iteration: 2  Process: /tb_axi_to_axi_lite/i_dut/i_axi_to_axi_lite/i_axi_burst_splitter/i_axi_burst_splitter_ar_chan/i_axi_burst_splitter_counters/#ASSIGN#545 File: /scratch/msc19f10/axi/src/axi_burst_splitter.sv Line: 545
```

accuminium, 2021-01-20:
This can probably be fixed by removing the `'x` assignments in [the `id_queue` module](https://github.com/pulp-platform/common_cells/blob/v1.20.0/src/id_queue.sv).  Can you try replacing all `'x` in the `id_queue` with `'0`?  If that works, the fix can be upstreamed.

(A bit of background: The [SV coding style guide](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md) we follow used to suggest `'x` for invalid values. However, this causes many problems in different tools and [has since been revised to the opposite](https://github.com/lowRISC/style-guides/commit/4e25caa9a7d6ae54a96e4c078e0a04ccf4f5ae5e#diff-a3f54e4bd7591bd9e5f2166742361599c6bca5747158c3b9dcc1bf1e8ab60ddf).)

WRoenninger, 2021-01-20:
Changing all instances inside `id_queue` where ther in an `'x` assignment as default/reset indeed resolves the issue.

There is a pull request ongoing for the `id_queue` that introduces simultanous push/pop (https://github.com/pulp-platform/common_cells/pull/106). Would be probably the best to introduce that change there?

accuminium, 2021-01-20:
> Changing all instances inside `id_queue` where there ia an `'x` assignment (as default/reset value) indeed resolves the issue.

Great!

> There is a pull request ongoing for the `id_queue` that introduces simultaneous push/pop ([pulp-platform/common_cells#106](https://github.com/pulp-platform/common_cells/pull/106)). Would be probably the best to introduce that change there?

Ah yes, thanks for reminding me of that one. But please open a separate PR based on the current `master` that contains only this fix. I can then rebase the other PR on this fix.

commit e79dfcdd (2021-01-21): Update `common_cells` to 1.20.1 to fix #150
PR #156 (2021-01-21): Update `common_cells` to 1.20.1 to fix #150
Fixes #150.
commit ac5998df (2021-01-21): Merge pull request #156 from pulp-platform/update-common_cells

Update `common_cells` to 1.20.1 to fix #150
closed 2021-01-21 by accuminium