pulp-platform/axi #153: Harmonize ports and parameters

accuminium, 2021-01-20:
Let us collect the changes required to harmonize ports and parameters and to minimize incompatibilities with EDA tools.  Those changes will be breaking (as in "backwards-incompatible"), so let us make sure we get them right.  **This is currently a draft** and contributors are kindly asked to comment.

Preamble: The key words "MUST", "MUST NOT", "REQUIRED", "SHALL", "SHALL NOT", "SHOULD", "SHOULD NOT", "RECOMMENDED", "NOT RECOMMENDED", "MAY", and "OPTIONAL" in this document are to be interpreted as described in [BCP 14](https://tools.ietf.org/html/bcp14) [[RFC2119]](https://tools.ietf.org/html/rfc2119) [[RFC8174]](https://tools.ietf.org/html/rfc8174) when, and only when, they appear in all capitals, as shown here.

## Parameters

### Legal Types

Every `parameter` of a synthesizable `module` MUST be either:
(a) a `type`, or
(b) a (vector of) one of the following SystemVerilog types:
  - `bit` or `logic`, which MAY be `signed` but by default are `unsigned`, or
  - `byte`, `shortint`, `int`, or `longint`, which MUST be followed by `unsigned` or `signed`, or

(c) a `typedef`ed type of one of the types in (b).

In particular, `struct`s and `string`s MUST NOT be used as `parameter` of a `module`.

Rationale: Many tools do not properly implement complex types for parameters.

For non-synthesizable `module`s and `class`es, the key words MUST and MUST NOT in this section are relaxed to SHOULD and SHOULD NOT, respectively.  (In particular, testbench `module`s MAY use `time` and `string` parameters.)

### Signedness

If an integer parameter (i.e., `byte`, `shortint`, `int`, or `longint`) is not supposed to take negative values, it MUST be declared `unsigned` instead of `signed`.

### Default Value

Every `parameter` MUST have a default value.

If possible, the default value SHOULD be a *null* value that is outside the legal range of the parameter (e.g., a signal width of zero).  In this case, the `module` SHOULD contain an assertion to ensure that the parameter is set to a value other than the *null* value at instantiation.

Rationale: Many tools require `parameter`s to have a default value, but in many cases a `parameter` that is not set at instantiation indicates an error that should be detected. 

### Derived Parameters

The parameter list of a `module` MUST NOT contain `localparam`s.  Rationale: Unsupported by some tools.

Instead, if the value of a parameter is derived from another parameter and should not be overridden at instantiation, the line above the derived parameter SHOULD be as follows:
```sv
/// Dependent parameter, DO NOT OVERRIDE!
```

### Names

- All `parameter` names MUST be UpperCamelCase.
  Rationale: [style guide](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#constants).
- Parameters MUST NOT be prefixed with `Axi`.
  Example: A module with a parametrizable data width has a `parameter` named `DataWidth`, not ~`AxiDataWidth`~.
  Rationale: Every module name starts with `axi_` and prefixing `parameter`s with `Axi` is redundant.
- When a `parameter` only applies to one port, it MUST start with the prefix of the port (converted to UpperCamelCase).
  Example: For a crossbar, the ID width at its `slv_ports` would be given by a `parameter` named `SlvPortsIdWidth`.
- When a `parameter` defines the number of bits in a signal, it SHOULD end with `Width`.
- When a `parameter` defines a quantity other than bits in a signal, it SHOULD start with `Num`.  `No` MUST NOT be used as prefix for a quantity `parameter`.  (Rationale: easily mistaken for negation, e.g., "no registers").

## Ports

TODO

accuminium, 2021-01-20:
RFC CC @zarubaf @fabianschuiki @WRoenninger

zarubaf, 2021-01-21:
Thanks for putting this together. This looks very, very promising. There is currently only one thing which came to my mind:

Regarding the interface wrapper for multiple slave or master ports this is essentially an unpacked type where the style guide recommends [big-endianess](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#summary-3). So I would recommend:
```systemverilog
axi_if.slv_port slv_ports [NumSlvPorts],
axi_if.mst_port mst_ports [NumMstPorts],
```
or 
```systemverilog
axi_if.slv_port slv_ports [0:NumSlvPorts-1],
axi_if.mst_port mst_ports [0:NumMstPorts-1],
```

accuminium, 2021-01-21:
> Regarding the interface wrapper for multiple slave or master ports this is essentially an unpacked type where the style guide recommends [...]

Good point, thanks! I am adding it.  To precisely follow the style guide, let us omit the space between the signal name and the array.

WRoenninger, 2021-01-21:
Thanks for defining these. They look very well thought out. I will update my PRs #116  #115 and #33 accordingly when these harmonization guidelines are finalized.

I could do the flatening of the `axi_pkg::xbar_cfg_t` struct for `axi_xbar` as part of #116 as I have added there doc descriptions for the parameters there already. 

One question regading #33. There the LLC has a master and slave port with AXI4+ATOP and an AXI4 Lite cfg slave port. So the names should be therefore:
```systemverilog
// Slave Port facing CPU
input  slv_port_axi_req_t slv_port_req_i,
output slv_port_axi_rsp_t slv_port_rsp_o,
// Master Port facing Memory
output mst_port_axi_req_t mst_port_req_o,
input  mst_port_axi_rsp_t mst_port_rsp_i,
// Configuration Lite Slave port
input  axi_lite_req_t cfg_slv_port_req_i,
output axi_lite_rsp_t cfg_slv_port_rsp_o, 
```
Also #33 would require some additional work from my part as the LLC submodules use a cfg struct to propagate the LLC configuration.

accuminium, 2021-01-22:
> Thanks for defining these. They look very well thought out. I will update my PRs #116 #115 and #33 accordingly when these harmonization guidelines are finalized.

Great, thanks!

> I could do the flattening of the `axi_pkg::xbar_cfg_t` struct for `axi_xbar` as part of #116 as I have added there doc descriptions for the parameters there already.

I am fine with flattening the parameters of the XBAR in the same PR as adding the docstrings, but please do not mix this up with the crossbar-internal pipeline change.  The crossbar pipeline change is complex enough in itself, and I would like to keep it separate for now.  (The changes for pipelining the crossbar need careful reviewing to make sure we still guarantee deadlock freedom.  Reviewing that is not high priority for me at the moment, because when a crossbar complex enough that it needs to be pipelined inside, it has many master and slave ports, and *internal* pipelining would add a lot of area -- which again has implications on timing.  So I do not think there is urgent need for making the crossbar internally pipelineable.)  You do not have to keep the `xbar_pipeline` branch rebased on `master` with high priority; instead please refactor all changes that are not related to pipelining and the `axi_demux` FIFO->counter into a different PR.

> One question regarding #33. There the LLC has a master and slave port with AXI4+ATOP and an AXI4 Lite cfg slave port. So the names should be therefore:
> 
> ```systemverilog
> // Slave Port facing CPU
> input  slv_port_axi_req_t slv_port_req_i,
> output slv_port_axi_rsp_t slv_port_rsp_o,
> // Master Port facing Memory
> output mst_port_axi_req_t mst_port_req_o,
> input  mst_port_axi_rsp_t mst_port_rsp_i,
> // Configuration Lite Slave port
> input  axi_lite_req_t cfg_slv_port_req_i,
> output axi_lite_rsp_t cfg_slv_port_rsp_o, 
> ```

I agree except for the configuration port:

```sv
input  cfg_slv_port_axi_lite_req_t cfg_slv_port_req_i,
output cfg_slv_port_axi_lite_rsp_t cfg_slv_port_rsp_o,
```
because "If a parameter only applies to one port, it MUST start with the prefix of the port [...]".  I realize this is a bit bulky to write, but it helps preventing type collisions.

> Also #33 would require some additional work from my part as the LLC submodules use a cfg struct to propagate the LLC configuration.

Yes.

WRoenninger, 2021-01-22:
Thank you for the clarification. 

Just a heads up, the changes for the `xbar_pipeline` are required for the LLC. There is a bypass for uncached memory accesses using both an `axi_demux` and `axi_mux`. The issue is the same deadlocking that happens if you put pipeline stages into the cross of the crossbar currently. The LLC also acts as sort of a 'pipeline' and needs the fix from #116 to allow simultaneous LLC and bypass write accesses.

accuminium, 2021-01-22:
> Just a heads up, the changes for the `xbar_pipeline` are required for the LLC. There is a bypass for uncached memory accesses using both an `axi_demux` and `axi_mux`. The issue is the same [...]

I see, thanks for clarifying this.  This gives the changes to `axi_demux` higher priority again.

WRoenninger, 2021-01-25:
> A crossbar with multiple slave and master ports would contain in its inputs and outputs:
> 
> ```systemverilog
> // Slave Ports
> input  slv_ports_axi_req_t [NumSlvPorts-1:0] slv_ports_req_i,
> output slv_ports_axi_rsp_t [NumSlvPorts-1:0] slv_ports_rsp_o,
> // Master Ports
> output mst_ports_axi_req_t [NumMstPorts-1:0] mst_ports_req_o,
> input  mst_ports_axi_rsp_t [NumMstPorts-1:0] mst_ports_rsp_i,
> ```

Should the req/rsp type in this case not be singular on the port part?
Rationale: The types themselves describe a single request/response struct of an individual port. The construct that makes the port plural is the array definition directly in the module port definition. So the proposition is to change it to:

```systemverilog
// Slave Ports
input  slv_port_axi_req_t [NumSlvPorts-1:0] slv_ports_req_i,
output slv_port_axi_rsp_t [NumSlvPorts-1:0] slv_ports_rsp_o,
// Master Ports
output mst_port_axi_req_t [NumMstPorts-1:0] mst_ports_req_o,
input  mst_port_axi_rsp_t [NumMstPorts-1:0] mst_ports_rsp_i,
```

Dropping the *s* from the parameterized type, however keep it in the port name.

Consequentially: *_ports_axi_req_t would be the type with the array included:
```systemverilog
typedef slv_port_axi_req_t [NumSlvPorts-1:0] slv_ports_req_t;
```

accuminium, 2021-01-25:
> Should the req/rsp type in this case not be singular on the port part?
> Rationale: The types themselves describe a single request/response struct of an individual port. The construct that makes the [...]

I agree that it is more intuitive to use a singular name for the type for an array port.  However, I am not convinced that the additional complexity of the type naming rule is worth the gained intuitiveness.

Currently, the rule is
> If a `parameter` only applies to one port, its name MUST start with the prefix of the port (converted to the casing dictated above) or with `Num` (see below) followed by the prefix of the port.

Should we change that to "If a `parameter` only applies to one port, its name MUST start with the prefix of the port (converted to the casing dictated above **and to singular if the port is an array**) or with `Num` (see below) followed by the prefix of the port." ?

> Consequentially: *_ports_axi_req_t would be the type with the array included:
> 
> ```systemverilog
> typedef slv_port_axi_req_t [NumSlvPorts-1:0] slv_ports_req_t;
> ```

`slv_ports_axi_req_t`, because an AXI request struct type MUST end with `axi_req_t`.

accuminium, 2021-02-01:
> Should we change that to "If a `parameter` only applies to one port, its name MUST start with the prefix of the port (converted to the casing dictated above **and to singular if the port is an array**) or with `Num` (see below) followed by the prefix of the port." ?

Updated! :+1:

accuminium, 2021-02-01:
Any other change requests or objections? If not, I would add this to our [Contribution Guidelines](https://github.com/pulp-platform/axi/blob/master/CONTRIBUTING.md) and start accepting PRs that implement these changes.

micprog, 2022-01-20:
>- A request type MUST end with `X_req_t`.
>- A response type MUST end with `X_rsp_t`.

Would it make sense to use `X_resp_t` for the response types? The main rationale for this is the already implemented typedef functions, although these could be changed as well: https://github.com/pulp-platform/axi/blob/20311e79ad573e4a973a792f5b102ee504a1d1c4/include/axi/typedef.svh#L118-L125

accuminium, 2022-01-20:
> > * A request type MUST end with `X_req_t`.
> > * A response type MUST end with `X_rsp_t`.
> 
> Would it make sense to use `X_resp_t` for the response types? The main rationale for this is the already implemented typedef functions, although these could be changed as well:

The rationale for `X_rsp_t` instead of `X_resp_t` is two-fold:
1) To clearly distinguish the response struct from the AXI response type (i.e., the `RESP` field).
2) To have symmetry in the length of the request and response struct type names (`X_req_t` and `X_rsp_t`).