pulp-platform/axi #167: A config example file for axi_intercon_gen.py

tunghoang290780, 2021-03-28:
Hello,

@accuminium 

Is there any config example file used as the input of axi_intercon_gen.py script? I try axi.core but it seems not working

Thanks!

accuminium, 2021-03-29:
Hi @tunghoang290780,

Thanks for your question.

[`axi.core` contains an example configuration for `axi_intercon_gen`](https://github.com/pulp-platform/axi/blob/v0.27.1/axi.core#L83-L100).  Have you tried that?  If yes and it does not work, what errors do you get?

tunghoang290780, 2021-03-29:
Hi @accuminium 

I hope my running command is correct `./scripts/axi_intercon_gen.py axi.core`  I got errors as below

[image]

accuminium, 2021-03-30:
`axi.core` is the Manifest file for FuseSoC, and it is not intended to be an input file to `scripts/axi_intercon_gen.py`.

Instead, you need to write a separate file that describes how your interconnect instance should be configured.  [`axi.core` contains an example configuration](https://github.com/pulp-platform/axi/blob/v0.27.1/axi.core#L83-L100).  Based on that example, you could write the following `my_xbar.yml` file:
```yml
vlnv: pulp-platform.org:axi:xbar:0.27.2
parameters:
  masters:
    dma:
      id_width : 1
    ibus:
      id_width : 2
  slaves:
    ram:
      offset : 0
      size: 0x10000000
    gpio:
      offset: 0x91000000
      size: 0x1000
    rom:
      offset : 0xffff0000
      size : 32768
```
Invoking `python3 scripts/axi_intercon_gen.py my_xbar.yml` will then create an `axi_intercon.v` in the current directory, which instantiates `axi_xbar` as you defined in `my_xbar.yml`.

accuminium, 2021-04-13:
Did this answer your question?  If not, please feel free to reopen the issue.

closed 2021-04-13 by accuminium