pulp-platform/axi #179: The entire directory "common_cells" appears to be missing

rlew-metastability, 2021-08-13:
The file:

    axi_burst_splitter.sv 
    located at (.../axi/src/ axi_burst_splitter.sv) in git repo branch "v0.29.1" (git hash 442ff33)

includes registers.svh on line 17 (`include "common_cells/registers.svh").  The entire subdirectory "common_cells" appears to be missing.

zarubaf, 2021-08-17:
We use [`bender`](https://github.com/pulp-platform/bender) to resolve the dependencies for us. There is a `Bender.yml` in all of the IPs of PULP Platform which mention the exact dependencies of each repository.

You can of course use the IP without `bender` but you will need to checkout the repositories and assemble the compile scripts on your own.

So my suggestion would be to quickly install `bender` which will do the tedious task for you :-P

closed 2021-08-17 by zarubaf

rlew-metastability, 2021-08-17:
Thank you for your quick response!  I wasn't aware of the Bender tool.  I'll check the .yml dependency file and learn how the Bender tool works.