pulp-platform/axi #181: what's the documentation tools?

qian-gu, 2021-08-19:
the auto-generated documentation looks awesome, what tools is this repo using please?

zarubaf, 2021-08-19:
Glad you like this!

--> https://github.com/zarubaf/morty 

We haven't really continued working on it for a while so in case you like it and also want to improve it, we would very much appreciate contributions :)

qian-gu, 2021-08-19:
Great Job! I have been looking for a tool like the sphinx or doxygen for Verilog/SV language, glad to contribtue to it if I can.

qian-gu, 2022-08-02:
Sorry for long time reopen this issue.

Morty is great and everything is okay except for `include` syntax in my design. So I try to build the doc of this repository on my local machine, as this repo also use the `include` syntax too:

```bash
bender sources -f > flatten.json
morty -f flatten.json --doc doc
```

and I got the error message:

```
Error: Failed to preprocess `/home/qian/repositories/axi/.bender/git/checkouts/common_cells-87388673ac9079ac/src/isochronous_4phase_handshake.sv`

Caused by:
    0: Include error
    1: File error: "common_cells/registers.svh"
    2: No such file or directory (os error 2)
```

I checked the generated `flatten.json` file and the `include_dirs` fields are empty which seems the reason of failure, but I have no idea how to solve it. Did I miss something or made some mistakes when using bender and morty? thanks for help.

andreaskurth, 2022-08-03:
To build the documentation through CI, the command is
https://github.com/pulp-platform/axi/blob/64ca21eb2ffcc26ed9b7635d19723fa227f38699/.github/workflows/doc.yml#L55-L56

Have you tried that?

qian-gu, 2022-08-05:
@andreaskurth problem solved, thanks for help.

closed 2022-08-05 by qian-gu