pulp-platform/axi #184: axi_dw_up/downsizer: Latch inference when synthesizing with Mentor Precision

WRoenninger, 2021-09-23:
When synthesizing with _Mentor Precision_, the tool infers latches in the `axi_dw_upsizer` and `axi_dw_downsizer`.
Seems the tool has an issue with the automatic variable `r_data` and `w_data` in both modules. 
Links to the of the offending codelines of `axi_dw_upsizer` and `axi_dw_downsizer`:
https://github.com/pulp-platform/axi/blob/f847575bab7269f88f6caa36f01275de2007378a/src/axi_dw_upsizer.sv#L478-L492
https://github.com/pulp-platform/axi/blob/f847575bab7269f88f6caa36f01275de2007378a/src/axi_dw_upsizer.sv#L595-L607
https://github.com/pulp-platform/axi/blob/f847575bab7269f88f6caa36f01275de2007378a/src/axi_dw_downsizer.sv#L550-L562
https://github.com/pulp-platform/axi/blob/f847575bab7269f88f6caa36f01275de2007378a/src/axi_dw_downsizer.sv#L722-L737

At least in `axi_dw_upsizer` the assignment of `r_data` should still be the default `'0`. So `r_data` is only _saving_ the initialization value, which is then conditionally overwritten.

Then however when synthesizing with _SYNOPSYS_ the latches are not there.

When adding an `else` statement to the `if` then the latch goes away at least in the first code snippet. So an idea for a fix from my side is to add the `else` condition and remove the automatic variable.

suehtamacv, 2021-09-28:
Hi @WRoenninger,

Thanks for reporting this. I do not have any experience with Precision, would you have perhaps an example analysis script for this tool? Bender also does not support that, so I would not be able to generate such an script easily.

Matheus

WRoenninger, 2021-09-29:
I have locally a patched bender version where I added the precision specific script. I was planning to make a PR for it.
The command to add files is:
```tcl
# Defines seem to be globally only
setup_design -defines {+define+TARGET_FPGA +define+TARGET_SYNTHESIS}

add_input_file -format SystemVerilog2012 -search_path {/abs/path/include1 /abs/path/include2} {/abs/path/src1.sv /abs/path/src2.sv}
```
Needs to be absolute paths as variable expansion in the tcl does sadly not really work.

suehtamacv, 2021-10-01:
Hi @WRoenninger 

I am looking into this. I cannot simply remove the automatic variable: in fact, I added this to avoid a compatibility issue with an old QuestaSim version: https://github.com/pulp-platform/axi/pull/124. 

I managed to reproduce your solution, and it does solve the latches, but I am working on generalizing this for the other four cases. But I would say that removing the automatic variable is not an option, which means this issue is a bit more involved than expected. 

Matheus

WRoenninger, 2021-10-03:
For bender and Precision see: https://github.com/pulp-platform/bender/pull/55

PR #185 (2021-11-11): Fix latch inference on the data width converters with the Mentor Precision synthesis tool
Closes #184.

This PR reworks a construct that would case latches to be infered on both the `axi_dw_downsizer` and the `axi_dw_upsizer`, when synthesizing them with the Mentor Precision.
closed 2021-11-11 by accuminium