pulp-platform/axi #194: AXI upsizer fix

luca-valente, 2022-01-05:
Hi @accuminium :smile: 

I think I spotted a bug in the `axi_dw_upsizer` when it is instantiated with `AxiMstPortDataWidth=128` and `AxiSlvPortDataWidth=64`. I'll try to describe it with much detail as possible. I spotted it when working on FPGA  and using gdb, so without much visibility.

I was connecting my logic to an AXIDDR4 controller provided by Xilinx. My design exposes an AXI Master port with `AXI_DATA_WIDTH=64` while the default `AXI_DATA_WIDTH` of the AXI Slave port of the Xilinx IP is `128`. That is why I inserted the `axi_dw_upsizer`, but this led to the error I'll describe below. 

When changing the `AXI_DATA_WIDTH` value of the Xilinx IP to `64`, and removing the upsizer, the bug disappears. That is why I believe there is a bug in the `axi_dw_upsizer`.

With `AXI_DATA_WIDTH=128` the address (1-byte words) layout for 2 `ax_beat`s is the following:
```
beat 0:   f  e  d  c  b  a  9  8  7  6  5  4  3  2  1  0
beat 1:  1f 1e 1d 1c 1b 1a 19 18 17 16 15 14 13 12 11 10
```
When writing in the `64 LSB`, everything is fine. While, when writing in the `64 MSB`, the resulting address coming out of the upsizer is subtracted by 10. Hence the layout is the following:
 ```
beat 0:  1f 1e 1d 1c 1b 1a 19 18  7  6  5  4  3  2  1  0
beat 1:  2f 2e 2d 2c 2b 2a 29 28 17 16 15 14 13 12 11 10
```
The risc-v debug module writes 32-bits words at a time, not 128 or 64.  When writing to `0x0` or `0x4` the upsizer works fine. When writing to `0x18` with `gdb`, the write actually goes @ `0x08` and so on: with `z>0x08`, `y*0x10 + 0x0z` becomes `(y-1)*0x10 + 0x0z`.

If I had found this issue while simulating in Questasim I would have debugged deeper, but on FPGA this was the best I could do :smile: 

Cheers,
Luca

accuminium, 2022-01-05:
Hi @luca-valente, thanks for reporting the problem you are observing.

I tried to reproduce it, but I could not. Here is the [test case](https://github.com/pulp-platform/axi/blob/194-axi-upsizer-problem/test/tb_axi_dw_upsizer_1.sv#L150-L157) I created:
```sv
        aw.ax_addr = 64'h18;
        aw.ax_size = 3'd2;
        aw.ax_burst = axi_pkg::BURST_INCR;
        master_drv.send_aw(aw);
        w.w_data = 64'h07060504_03020100;
        w.w_strb = 8'b0000_1111;
        w.w_last = 1'b1;
        master_drv.send_w(w);
```
It sends a 4-byte word (`size = 3'd2`) to address `64'h18`. The 64-bit W beat contains the data in the lower 4 byte (since `64'h18` is 64-bit aligned). The upsizer then outputs a 32-bit write to `64'h18` and
```
data: 128'h00000000_03020100_00000000_00000000
strb: 16'b0000_1111_0000_0000
```
which is correct.

Can you please list both writes that you execute through the debug module to get the two beats you listed?

luca-valente, 2022-01-05:
Hi,
when I have the upsizer and use gdb I have the following outcome:
```
(gdb) x/20x 0x80000000
0x80000000 <_start>:    0x41014081      0x42014181      0x47014681      0x48014781
0x80000010 <_start+16>: 0x45014481      0x46014581      0x4b014a81      0x4c014b81
0x80000020 <_start+32>: 0x49014881      0x4a014981      0x4f014e81      0x62f94f81
0x80000030 <_start+48>: 0x4d014c81      0x4e014d81      0x0002d863      0x12974505
0x80000040 <_start+64>: 0x3002a073      0x02fe4285      0x00000297      0x09028293
(gdb) set *( (int *) 0x80000018 ) = 0xa5a5a5a5
(gdb) x/20x 0x80000000
0x80000000 <_start>:    0x41014081      0x42014181      0xa5a5a5a5      0x48014781
0x80000010 <_start+16>: 0x45014481      0x46014581      0x4b014a81      0x4c014b81
0x80000020 <_start+32>: 0x49014881      0x4a014981      0x4f014e81      0x62f94f81
0x80000030 <_start+48>: 0x4d014c81      0x4e014d81      0x0002d863      0x12974505
0x80000040 <_start+64>: 0x3002a073      0x02fe4285      0x00000297      0x09028293
(gdb) set *( (int *) 0x80000028 ) = 0xabbaabba
(gdb) x/20x 0x80000000
0x80000000 <_start>:    0x41014081      0x42014181      0xa5a5a5a5      0x48014781
0x80000010 <_start+16>: 0x45014481      0x46014581      0xabbaabba      0x4c014b81
0x80000020 <_start+32>: 0x49014881      0x4a014981      0x4f014e81      0x62f94f81
0x80000030 <_start+48>: 0x4d014c81      0x4e014d81      0x0002d863      0x12974505
0x80000040 <_start+64>: 0x3002a073      0x02fe4285      0x00000297      0x09028293
(gdb) 
```
Without the upsizer, everything works fine:
```(gdb) x/20x 0x80000000
0x80000000 <_start>:    0x41014081      0x42014181      0x43014281      0x44014381
0x80000010 <_start+16>: 0x45014481      0x46014581      0x47014681      0x48014781
0x80000020 <_start+32>: 0x49014881      0x4a014981      0x4b014a81      0x4c014b81
0x80000030 <_start+48>: 0x4d014c81      0x4e014d81      0x4f014e81      0x62f94f81
0x80000040 <_start+64>: 0x3002a073      0x02fe4285      0x0002d863      0x12974505
(gdb) set *( (int *) 0x80000018 ) = 0xabbaabba 
(gdb) x/20x 0x80000000
0x80000000 <_start>:    0x41014081      0x42014181      0x43014281      0x44014381
0x80000010 <_start+16>: 0x45014481      0x46014581      0xabbaabba      0x48014781
0x80000020 <_start+32>: 0x49014881      0x4a014981      0x4b014a81      0x4c014b81
0x80000030 <_start+48>: 0x4d014c81      0x4e014d81      0x4f014e81      0x62f94f81
0x80000040 <_start+64>: 0x3002a073      0x02fe4285      0x0002d863      0x12974505
(gdb) set *( (int *) 0x80000028 ) = 0xa5a5a5a5 
(gdb) x/20x 0x80000000
0x80000000 <_start>:    0x41014081      0x42014181      0x43014281      0x44014381
0x80000010 <_start+16>: 0x45014481      0x46014581      0xabbaabba      0x48014781
0x80000020 <_start+32>: 0x49014881      0x4a014981      0xa5a5a5a5      0x4c014b81
0x80000030 <_start+48>: 0x4d014c81      0x4e014d81      0x4f014e81      0x62f94f81
0x80000040 <_start+64>: 0x3002a073      0x02fe4285      0x0002d863      0x12974505
(gdb) 
```
The `AXI_ADDRESS_WIDTH` is 29, so the 2 MSB are ignored. 
What do you think? Maybe the Xilinx IP is not fully AXI compliant?

Luca

luca-valente, 2022-01-06:
Hi @accuminium 
I guess I can easily do the second option!
I'll update you as soon as I'll run the test! :smile:

luca-valente, 2022-01-11:
Hi @accuminium 
I simulated the code load through JTAG in RTL, as it happens for the FPGA. Looking at the waveforms, I see no errors on the upsizer side. One thing I was wrong about: the data width of the requests coming from the debug module is actually 64, not 32.
 Anyway, when the transaction is the following (using the same code as the `axi_tb`)
```
        aw.ax_addr = 64'h18;
        aw.ax_size = 3'd3;
        aw.ax_burst = axi_pkg::BURST_INCR;
        master_drv.send_aw(aw);
        w.w_data = 64'h07060504_03020100;
        w.w_strb = 8'b1111_1111;
        w.w_last = 1'b1;
        master_drv.send_w(w);
```
On the 128b master port, we have:
```
        aw.ax_addr = 64'h18;
        aw.ax_size = 3'd3;
        aw.ax_burst = axi_pkg::BURST_INCR;
        master_drv.send_aw(aw);
        w.w_data = 128'h07060504_03020100_00000000_00000000;
        w.w_strb = 16'b1111_1111_0000_0000;
        w.w_last = 1'b1;
        master_drv.send_w(w);
```
Which is perfectly fine. Hence, I believe there is some sort of incompatibility of the Xilinx IP with writes for the second case.
Do you think it is worth investigating further?

accuminium, 2022-01-11:
Thanks for the feedback, @luca-valente. I agree that this conversion by the upsizer is correct. Could you please also verify this on the FPGA by inserting one ILA on the signals described above at the interface of the DDR4 controller?

I looked through the [Xilinx DDR4 Known Issues](https://support.xilinx.com/s/article/69035?language=en_US), and I could not find anything related to what you are observing. Before contacting Xilinx support, we need the transactions at the interface of the DDR4 controller, so (1) we know the fault is certainly not in our circuit and (2) we can provide them with a precise I/O description of the problem.

luca-valente, 2022-01-11:
Sure, I can do that :smile:

accuminium, 2022-02-10:
@luca-valente: In your WR on 2022-01-29, you wrote "there were no hardware bugs and [...] the DDR4 Xilinx is working as supposed to". Does that mean that you now have an implementation with correct behavior? If yes, what made the difference?

luca-valente, 2022-02-11:
Ciao @accuminium 
it works without the upsizer. With the upsizer, the error still stands.

accuminium, 2022-02-11:
I see, thanks for the clarification. When you find some time, it would be very helpful to debug this by seeing the ILA-probed signals before and after the upsizer and before the DDR4 controller.