pulp-platform/axi #195: axi_dw_downsizer W channel

Realhssuperman, 2022-01-10:
https://github.com/pulp-platform/axi/blob/5380c5fc99aed848d59d0fd8867e461c7b24fd4d/src/axi_dw_downsizer.sv#L729

Hi Folks,

an above link tells that, it is able to manipulate **slv_req_i.w.data** w/o **slv_resp_o.w_ready**. 

On the other hand, in the case of **axi_dw_downsizer**,  it is able to serialize incoming data only after check both **slv_resp_o.w_ready && slv_req_i.w_valid**.

accuminium, 2022-01-10:
Hello @Realhssuperman, thanks for getting in touch with us.

> an above link tells that, it is able to manipulate `slv_req_i.w.data` w/o `slv_resp_o.w_ready`.

`slv_req_i.w.data` is valid as soon as `slv_req_i.w_valid` is high. Thus, `axi_dw_downsizer` can start processing 
the data. As soon as `axi_dw_downsizer` has completed processing the data, it sets `slv_resp_o.w_ready` high for one clock cycle.

> On the other hand, in the case of `axi_dw_downsizer`, it is able to serialize incoming data only after check both `slv_resp_o.w_ready && slv_req_i.w_valid`.

In each clock cycle with `slv_req_i.w_valid && slv_resp_o.w_ready`, one W beat on the slave port is processed.

So I think your understanding is correct, but there is nothing wrong with it. Do you think there is a problem? Or do you have a follow-up question?

Realhssuperman, 2022-01-10:
Hello, @accuminium , thanks for your fast response.

Sorry, I just correct my first comment likes, ' On the other hand, in the case of `axi_dw_upsize`' 

I think `axi_dw_upsize` looks reasonable since it takes w.data depending on both **ready & valid**.
https://github.com/pulp-platform/axi/blob/5380c5fc99aed848d59d0fd8867e461c7b24fd4d/src/axi_dw_upsizer.sv#L602

According to your comment, I think `axi_dw_downsizer` acknowledge that the incoming `slv_req_i.w.data` is legal, w/0 assert `slv_resp_o.w_ready` high, then make it high after manipulate `slv_req_i.w.data` in order to notice toward source, that `slv_req_i.w.data` has been processed.

But it seems like breaking **the principle of decoupled ready & valid**. 

1. You checked `slv_req_i.w_valid`
2. It allows `mst_req.w_valid = 1'b1`  in Line #734
3. It is a part of trigger `if (mst_resp.w_ready && mst_req.w_valid)` in Line #750
4. Then update `w_req_d.aw.addr` Line #756
5. It is a part of trigger `if (w_req_q.burst_len == 0 || (aligned_addr(w_req_d.aw.addr, w_req_q.orig_aw_size) != aligned_addr(w_req_q.aw.addr, w_req_q.orig_aw_size)))` Link #768, to assert `slv_resp_o.w_ready` high.

Ultimately, `slv_req_i.w_valid` **combines** with `slv_resp_o.w_ready` , which breaks the decoupled principles.

Realhssuperman, 2022-01-10:
@accuminium, 

Just got that **the principle of decoupled ready & valid** means that ready & valid is combined without F/F in the middle. 

https://github.com/pulp-platform/axi/blob/5380c5fc99aed848d59d0fd8867e461c7b24fd4d/src/axi_dw_downsizer.sv#L768

This tells `slv_resp_o.w_ready` will assert high only after 'D=>Q' update, which means it must have sequential blocks in the middle.

accuminium, 2022-01-10:
That is correct, there is a combinatorial path between `slv_req_i.w_valid` and `slv_resp_o.w_ready` in `axi_dw_downsizer`. (That module is not the only one in this repository that has a combinatorial path between the handshake signals; see below.)

This deviates from A3.1.1 of the AXI specification, which says:
> All output signal changes can only occur after the rising edge of [the clock]. On Manager [(Master in the original spec)] and Subordinate [(Slave in the original spec)] interfaces, there must be no combinatorial paths between input and output signals.

This is what you refer to as *principle of decoupled ready & valid*, right?

If so, I think there are three main aspects to this:

### Why did we implement it this way?

Without a combinatorial path between `valid` and `ready`, a handshake can follow one of two patterns:
1. The downstream module has `ready` low after reset. In the clock cycle *after* a `valid`, it sets `ready` high. This forces the transfer to take at least two cycles, which is not optimal.
2. The downstream module has `ready` high after reset. It thus must be able to unconditionally accept a beat as soon as the upstream module set `valid` high. In full generality, not every module can do this for every beat, so many modules would need registers to buffer all bits of a beat in order to implement this *default ready* scheme.

With a combinatorial path between `valid` and `ready`, the downstream module can react to `valid` in the same clock cycle and does not need registers to buffer a `valid` beat that it cannot immediately accept. Importantly, the combinatorial path only goes from valid to ready, never from ready to valid. This is compliant with the AXI dependency rules for handshake signals (A3.3.1):
> - The `valid` signal of the AXI interface sending information must not be dependent on the `ready` signal of the AXI interface receiving that information.
> - An AXI interface that is receiving information can wait until it detects a `valid` signal before it asserts its corresponding `ready` signal.

Our implementation is thus **not** prone to combinatorial loops or deadlocks on the handshake signals.

### What are potential issues with this implementation?

We implement the handshake protocol as a [Mealy FSM instead of a Moore FSM](https://en.wikipedia.org/wiki/Mealy_machine#Comparison_of_Mealy_machines_and_Moore_machines). Therefore, there are combinatorial I/O paths for some modules, which need to be taken into consideration for timing closure. In particular, when such a module is instantiated at the boundary of timing analysis, proper input and output delay constraints have to be set. Without that, timing may fail to converge/close. However, **static timing analysis (STA) will detect this issue**, so you do not risk taping out a circuit with timing violations that you did not notice.

We have not encountered any other issues in the multiple years and dozens of chips we have taped-out with our AXI modules. If you are aware of any other potential issue or have concerns, please let us know!

### What if I need strict specification compliance?

Since this is a deviation from the AXI specification and we aim for full compliance, we should state this in the ReadMe and explain how to resolve it. Fortunately, the solution is relatively simple: We will create an AXI module that adds a [`fall_through_register`](https://github.com/pulp-platform/common_cells/blob/v1.23.0/src/fall_through_register.sv) on each channel. The `fall_through_register` was designed precisely for the purpose of a *default ready* valid/ready handshake interface. You can then instantiate this new module in front of the existing module for which you need strict specification compliance.

In the long term, we might add a parameter to each module to instantiate the fall-through register internally. As this would require going through the ports of all modules and adding them where necessary, we would approach this incrementally and welcome contributions.

Realhssuperman, 2022-01-12:
> This is what you refer to as _principle of decoupled ready & valid_, right?

Absolutely.
 
> ### Why did we implement it this way?
> Without a combinatorial path between `valid` and `ready`, a handshake can follow one of two patterns:
> 
> 1. The downstream module has `ready` low after reset. In the clock cycle _after_ a `valid`, it sets `ready` high. This forces the transfer to take at least two cycles, which is not optimal.
> 2. The downstream module has `ready` high after reset. It thus must be able to unconditionally accept a beat as soon as the upstream module set `valid` high. In full generality, not every module can do this for every beat, so many modules would need registers to buffer all bits of a beat in order to implement this _default ready_ scheme. This would add many registers with no actual use.

I assume that 2nd pattern should take more costs to buffer, especially for the `axi_dw_downsizer.sv`, as not taken in your implementation. But in the case of upsizer, taking 2nd pattern leverages 1st pattern, although `axi_dw_upsizer.sv`  doesn't have buffer technically.

> ### What are potential issues with this implementation?
> We implement the handshake protocol as a [Mealy FSM instead of a Moore [...]

Some other design platforms name that `fall_through_register`  as `repeater`, given a boolean repeat, copy & repeat with repeat asserts high, pass with repeat assert low. It seems likes almost equivalent in many ways.

accuminium, 2022-01-25:
I would like to keep this issue open until we have implemented the solution described above, if you don't mind.