pulp-platform/axi #202: Questasim 2021.2 & axi_lite_demux & segfault

andreaskuster, 2022-01-29:
The `axi_demux` module has a known incompatibility with `Questasim 2020.2/2021.1` that has been resolved with the definition `TARGET_VSIM`, see [src/axi_demux.sv#L151](https://github.com/pulp-platform/axi/blob/master/src/axi_demux.sv#L151)

Since the error message/crash report for this problem is very cryptic (see error log below: `Fatal: Unexpected signal: 11.`), I actually ended up at the note of the questasim incompatibility after spending quite some time debugging it.

My suggestion is to check if `QUESTA` is definied instead, since this definition is automatically set for Questasim simulation instead of `TARGET_VSIM`, which is PULP specific as far as I know. Another option is to use the flattened structure that works for every EDA tool instead (similar as in [src/axi_lite_demux.sv#L287](https://github.com/pulp-platform/axi/blob/master/src/axi_lite_demux.sv#L287)).

Let me know which version you prefer and I will open a PR:

1. Use the flattened struct for everything i.e. `typedef logic [$bits(aw_chan_select_t)-1:0] aw_chan_select_flat_t;`
2. Replace `ifdef TARGET_VSIM` with `ifdef QUESTA`
3. Make both defines co-exist, i.e. 
```
`ifdef QUESTA
   `define TARGET_VSIM
`endif
```

Error log:
```
INFO     cocotb:simulator.py:267 # vsim -onfinish exit -pli "/home/andreas/PycharmProjects/axi-io-pmp/venv/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_modelsim.so" dut.dut -gRUSER_WIDTH=1 -gBUSER_WIDTH=1 -gARUSER_ENABLE=0 -gADDR_WIDTH=64 -gSTRB_WIDTH=8 -gID_WIDTH=8 -gWUSER_WIDTH=1 -gWAVES=1 -gREG_TYPE=1 -gDATA_WIDTH=64 -gARUSER_WIDTH=1 -gWUSER_ENABLE=0 -gAWUSER_ENABLE=0 -gAWUSER_WIDTH=1 -gRUSER_ENABLE=0 -gBUSER_ENABLE=0
INFO     cocotb:simulator.py:267 # Start time: 11:50:09 on Jan 28,2022
INFO     cocotb:simulator.py:267 # ** Note: (vsim-3812) Design is being optimized...
INFO     cocotb:simulator.py:267 # ** Warning: (vsim-3865) Due to PLI being present, global +acc is being enabled automatically. This will cause your simulation to run very slowly. Please use vsim -no_autoacc to disable this feature. This feature is now deprecated and will be removed from future releases.
INFO     cocotb:simulator.py:267 # vopt_stacktrace.vstf written
INFO     cocotb:simulator.py:267 # ** Fatal: Unexpected signal: 11.
INFO     cocotb:simulator.py:267 # /home/andreas/PycharmProjects/axi-io-pmp/src/pmp/pmp_entry.sv(1): Vopt Compiler exiting
INFO     cocotb:simulator.py:267 # ** Note: (vsim-12126) Error and warning message counts have been restored: Errors=1, Warnings=1.
INFO     cocotb:simulator.py:267 # Error loading design
INFO     cocotb:simulator.py:267 Error loading design
INFO     cocotb:simulator.py:267 # End time: 11:50:10 on Jan 28,2022, Elapsed time: 0:00:01
INFO     cocotb:simulator.py:267 # Errors: 1, Warnings: 1
ERROR    cocotb:simulator.py:271 Command terminated with error 12
========================================================================================================================================================================================== short test summary info ==========================================================================================================================================================================================
FAILED tb/test_axi_io_pmp.py::test_axi_io_pmp[questa-64-64-1] - AssertionError: Simulation terminated abnormally. Results file not found.
```

accuminium, 2022-02-09:
Thanks for reporting this, @andreaskuster!

I think your solution suggestion 1 is preferable, because
- we use the `$bits(type)` construct already in `axi_lite_demux`, `axi_err_slv`, `axi_burst_splitter`, and `axi_cdc_{dst,src}` and have not encountered problems with it so far;
- it reduces tool-specific code.

Could you please open a PR to implement the change?

andreaskuster, 2022-02-09:
done

commit fe5af79f (2023-04-10): axi_demux: Use flat select signal independent of tool-specific defines (#204)

This fixes #202 and further improves compatibility with known EDA tool
bugs.