pulp-platform/axi #203: Manipulation with static parameters & stream data

Realhssuperman, 2022-02-03:
https://github.com/pulp-platform/axi/blob/18d68d04fa46dc970a460c1baaca71a4e15ee0ee/src/axi_dw_downsizer.sv#L827

To pull metadata for covert, we have to refer to both static parameters & stream data together, like above. Usually, static parameters keep for configuration & build, stream data for the actual process. It looks unnatural for me to combine those heterogeneous things.

Directly speaking, if it substitutes to Verilog, `AxiMstPortStrbWidth` has to be kept with reset as a specific value (not zero), which is not likable for synthesis, even combine with streaming `slv_req_i.aw.size`.

accuminium, 2022-02-09:
Thanks for this comment, @Realhssuperman. If I understand you correctly, you suggest to not use a synthesis parameter together with signals in one expression, correct?

If so, why is that problematic?

Realhssuperman, 2022-02-28:
Explicitly speaking, as we compare `axi_dw_downsizer.sv` with TileLink WidthWidget, AXI4 one has several static register that has to be kept with **reset** and Vdd&Ground. On the other hand, TL one doesn't need to keep static register, all of them are streamed, which looks more better for me.

But as I talked with physical side, it is acceptable to set static register value on bus adapter.

In addition, I can't pull any idea to convert width w/o setting port widths as static value on AMBA, which alternates @accuminium.

andreaskurth, 2022-03-30:
> AXI4 one has several static register that has to be kept with reset and Vdd&Ground

Yes, our data width downsizer includes registers to track the state of outstanding transactions. Those registers are reset with `rst_ni`. In a physical implementation, the registers will have to be connected to power and ground rails (like all cells). Additionally, our data width downsizer has synthesis-time constants, which should be physically implemented as tie-high / tie-low cells or in another way suitable for read-only data.

> On the other hand, TL one doesn't need to keep static register, all of them are streamed, which looks more better for me.

TileLink responses (D channel) include the metadata required for the data width conversion of responses. AXI responses (R channel) do not include that metadata. Therefore, AXI data width converters have to store that metadata internally for all outstanding transactions whereas TileLink data width converters (or *WidthWidget*s) do not. This is a consequence of the protocol and not of the implementation.

> In addition, I can't pull any idea to convert width w/o setting port widths as static value on AMBA, which alternates @accuminium.

The data width is a physical property of an AXI on-chip network (corresponding to the number of wires of the `data` signals). We pass this property as synthesis-time constants (one data width before the conversion and one data width after the conversion) to the data width converter.

andreaskurth, 2022-03-30:
I currently don't see an action item or further clarifications required on this topic and therefore will close this issue. If you disagree, please reopen and describe why. Thank you for your inputs!

closed 2022-03-30 by andreaskurth

Realhssuperman, 2022-03-31:
No objection!