pulp-platform/axi #211: Are Data width and Address width changeable?

87657361, 2022-02-20:
Hello, 

The default config of the data width is 64bit, and address width is 32bit, can I change the data width to 256bit, and address width to 48bit? if yes, how?

Thanks！

accuminium, 2022-02-21:
Hi @87657361, thanks for you question.

Yes, data width and address width of all modules are configurable. Each module has `parameter`s for this configuration. For most modules, those `parameter`s are called `AxiAddrWidth` and `AxiDataWidth`. The crossbar (`axi_xbar`) takes a [config `struct`](https://github.com/pulp-platform/axi/blob/v0.32.0/src/axi_pkg.sv#L394-L408) that contains those parameters.

accuminium, 2022-02-21:
> So the python script cannot config the data and address width, if wanna different data and address width, I need modify the code by myself, right?

Yes, the FuseSoC generator for the crossbar currently does not take the address and data width as parameters. If you would like to instantiate a crossbar with different widths that the default, you are welcome to extend the generator.

> And can I config the masters port and slave ports in different width? like, master data width 256bits, and data slave width 64bits

No, you have to instantiate a [data width converter](https://github.com/pulp-platform/axi/blob/master/src/axi_dw_converter.sv) to convert between any two data widths.

closed 2022-03-30 by andreaskurth

saberlililily, 2023-09-26:
> Hi @87657361, thanks for your question.
> 
> Yes, data width and address width of all modules are configurable. Each module has `parameter`s for this configuration. For most modules, those `parameter`s are called `AxiAddrWidth` and `AxiDataWidth`. The crossbar (`axi_xbar`) takes a [config `struct`](https://github.com/pulp-platform/axi/blob/v0.32.0/src/axi_pkg.sv#L394-L408) that contains those parameters.

Do I need to modify some other logic in your code when I change the data width and address width?