pulp-platform/axi #229: axi_xbar synthesized？

menglinhan, 2022-04-22:
Hello
Is the code of axi_xbar synthesized by genus? when using genus synthesize these code, which macro should I define?
thanks

andreaskurth, 2022-05-03:
Hi @menglinhan, thanks for your questions.

> Is the code of axi_xbar synthesized by genus?

`axi_xbar` contains synthesizable standard-compliant SystemVerilog code, so yes, Genus should correctly synthesize it. If you encounter problems, please report them here so we can investigate.

> when using genus synthesize these code, which macro should I define?

You should define `TARGET_SYNTHESIS` for all synthesis targets. If you use [Bender](https://github.com/pulp-platform/bender) to generate the Genus compilation TCL script (`bender script genus`), it will automatically be set. You will also have to provide the [dependencies at the correct version](https://github.com/pulp-platform/axi/blob/b5a6a68f602bdf5c73b4f18ce898b88987a37e33/Bender.yml#L11-L12). Again, using Bender automates this: Simply [download or install Bender](https://github.com/pulp-platform/bender#installation), open a shell where you have cloned this repository, execute `bender script genus > genus.tcl`, and run the `genus.tcl` script in Genus.

menglinhan, 2022-05-23:
Thanks for your reply, I will try it.

andreaskurth, 2022-06-26:
Sounds good. If you have problems, please let us know.

closed 2022-06-26 by andreaskurth