pulp-platform/axi #237: Docs on building the testbenches

vkomenda, 2022-06-17:
I would appreciate some info on how to build the testbenches using Vivado or verilator. It's not straightforward how to use the CI flow based on Quartus as a reference for that.

andreaskurth, 2022-07-06:
Thanks for your question, @vkomenda.

The testbenches contain non-synthesizable code, so they cannot be run in or directly translated to Verilator.

For Vivado, progress is being tracked in #226.

vkomenda, 2022-07-06:
> Thanks for your question, @vkomenda.
> 
> The testbenches contain non-synthesizable code, so they cannot be run in or directly translated to Verilator.
> 
> For Vivado, progress is being tracked in #226.

Thanks for letting know. If the verilator support is out of the question, please feel free to close this issue as a duplicate.

andreaskurth, 2022-07-06:
> If the verilator support is out of the question, please feel free to close this issue as a duplicate.

It is not out of question -- in fact, I think it would be very useful! -- but it is currently not work *we* plan to do in the foreseeable future. I would certainly be willing to review and eventually merge PRs that add Verilator-based TBs, though! I'd suggest to start with a simple module such as `axi_cut` to establish a working setup (that would also be part of CI) and a basic infrastructure, and to get first results relatively quickly. Full protocol checking in more complex modules, such as `axi_xbar` (or even its main components, `axi_demux` and `axi_mux`), is a lot more involved.