pulp-platform/axi #294: axi_cdc fpga implementation very inefficient

skokvermon, 2023-04-03:
Hi

axi_cdc uses fifos that don't infer dual SRAM. Fifo is implemented using registers, and it is very slow.
In my design, I had to lower fifo depth to 8 words in order to be able to run at 100mhz.

fifo_v3 also don't infer SRAM, which is very problematic.
I could succesfully use single and dual clock fifos from optimsoc which can infer xilinx SRAM (and probably altera too).
I think it would be nice to use such fifos for axi_cdc

omeag, 2024-03-04:
Hi sir, I've been reviewing the axi_cdc, but can't find the definitions for cdc_fifo_gray_src and cdc_fifo_gray_dst modules.
Could you kindly point me to where I can find them?

micprog, 2024-03-04:
Hi @skokvermon,
Generally, our IPs are focused on ASIC designs, not FPGA, thus we have not implemented inferring SRAMs, either for the CDC FIFOs or general FIFOs. Both of these use modules of the common_cells repository: [`fifo_v3`](https://github.com/pulp-platform/common_cells/blob/master/src/fifo_v3.sv) and [`cdc_fifo_gray'](https://github.com/pulp-platform/common_cells/blob/master/src/cdc_fifo_gray.sv).
The tool we use to collect dependencies and generate scripts for tools (bender) allows switching out of individual files, e.g. for FPGA synthesis. While we generally use FPGA to prototype designs we implement as ASICs and attempt to avoid significant divergence for proper testing, it may be good to have a more FPGA-targeted implementation of these IPs. Feel free to open a pull request to add these modules, I'd be happy to integrate them!

12yuuuu, 2026-11-04:
Hi, I made a memory-based version of the CDC FIFO and tested it on an Intel FPGA. I want to ask a few things before I open PRs.

The new cell `cc_cdc_fifo_gray_ram` has the same ports and parameters as `cc_cdc_fifo_gray`. The pointer CDC is the same code (it uses `cc_cdc_fifo_gray_src_ptr` / `_dst_ptr`). Only the storage is changed to a dual-clock memory with read enable, so FPGA tools can use block RAM. I also made `axi_cdc_ram`, same ports and parameters as `axi_cdc`, with five of the new FIFOs. The original modules are not changed.

Results on Arria 10 GX 160 (Quartus Pro 23.2, AXI 32/64/4/1, 300 MHz target on both clocks, DUT entity numbers):

| depth | module | ALMs | FFs | M20K | Fmax src / dst |
|---:|---|---:|---:|---:|---:|
| 4 | axi_cdc | 623 | 1,850 | 0 | 480 / 428 MHz |
| 4 | axi_cdc_ram | 55 | 130 | 8 | 549 / 554 MHz |
| 16 | axi_cdc | 1,991 | 5,398 | 0 | 390 / 395 MHz |
| 16 | axi_cdc_ram | 106 | 156 | 9 | 446 / 454 MHz |
| 64 | axi_cdc | 9,126 | 19,430 | 0 | 290 / 302 MHz |
| 64 | axi_cdc_ram | 153 | 216 | 9 | 442 / 409 MHz |

There are two differences, both written in the module header: the buffer is 2**LogDepth + 1 entries instead of + 2 (same data, same order), and the memory is not reset. One more note: at depth 4, Quartus keeps the small 4x7-bit B channel in registers. It still works, but that path needs a data-delay constraint.

For testing I added a `USE_RAM_DUT` switch to the upstream `cc_cdc_fifo_gray_tb` and ran it on Verilator and Questa. I also wrote extra FIFO tests (26 configs) and a five-channel transfer test for `axi_cdc_ram` (14 configs). All pass. I could not run `tb_axi_cdc` on the RAM version (no license here), and I did not do post-fit simulation or Xilinx.

Questions: (1) Which name do you prefer, `_ram`, `_mem` or `_fpga`? (2) I used the repo license header (ETH Zurich and University of Bologna, SHL-0.51) with my name as author. Is that OK? (3) I added new modules instead of a Bender file swap, because the capacity and reset are a little different. Is that OK?

If this is fine, I will open the common_cells PR first, then the axi PR. I used AI assistance (Claude) and will mark it in the commits.

12yuuuu, 2026-11-04:
Hi, I made a memory-based version of the CDC FIFO and tested it on an Intel FPGA. I want to ask a few things before I open PRs.

The new cell `cc_cdc_fifo_gray_ram` has the same ports and parameters as `cc_cdc_fifo_gray`. The pointer CDC is the same code (it uses `cc_cdc_fifo_gray_src_ptr` / `_dst_ptr`). Only the storage is changed to a dual-clock memory with read enable, so FPGA tools can use block RAM. I also made `axi_cdc_ram`, same ports and parameters as `axi_cdc`, with five of the new FIFOs. The original modules are not changed.

Results on Arria 10 GX 160 (Quartus Pro 23.2, AXI 32/64/4/1, 300 MHz target on both clocks, DUT entity numbers):

| depth | module | ALMs | FFs | M20K | Fmax src / dst |
|---:|---|---:|---:|---:|---:|
| 4 | axi_cdc | 623 | 1,850 | 0 | 480 / 428 MHz |
| 4 | axi_cdc_ram | 55 | 130 | 8 | 549 / 554 MHz |
| 16 | axi_cdc | 1,991 | 5,398 | 0 | 390 / 395 MHz |
| 16 | axi_cdc_ram | 106 | 156 | 9 | 446 / 454 MHz |
| 64 | axi_cdc | 9,126 | 19,430 | 0 | 290 / 302 MHz |
| 64 | axi_cdc_ram | 153 | 216 | 9 | 442 / 409 MHz |

There are two differences, both written in the module header: the buffer is 2**LogDepth + 1 entries instead of + 2 (same data, same order), and the memory is not reset. One more note: at depth 4, Quartus keeps the small 4x7-bit B channel in registers. It still works, but that path needs a data-delay constraint.

For testing I added a `USE_RAM_DUT` switch to the upstream `cc_cdc_fifo_gray_tb` and ran it on Verilator and Questa. I also wrote extra FIFO tests (26 configs) and a five-channel transfer test for `axi_cdc_ram` (14 configs). All pass. I could not run `tb_axi_cdc` on the RAM version (no license here), and I did not do post-fit simulation or Xilinx.

Questions: (1) Which name do you prefer, `_ram`, `_mem` or `_fpga`? (2) I used the repo license header (ETH Zurich and University of Bologna, SHL-0.51) with my name as author. Is that OK? (3) I added new modules instead of a Bender file swap, because the capacity and reset are a little different. Is that OK?

I opened the common_cells part as a draft PR: pulp-platform/common_cells#369. The axi PR will follow after that one. I used AI assistance (Claude) and marked it in the commit.