pulp-platform/axi #324: Logging not supported for Queue item error : AXI

ChandraLekha559, 2023-10-07:
I am trying to display the waveform of a sysytem verilog Queue for tb_axi_serializer  and got this error :
**# (vsim-4027) Logging is not supported for Queue item**: /tb_axi_serializer/ar_chan 

I am currently using **ModelSim SE-64 10.6** version.
Can someone please suggest a way to rectify this issue.

Thank you.

[image]

bluewww, 2023-10-27:
The message clearly says that this feature is not supported in your version of the simulator. Update the version of your simulator

ChandraLekha559, 2024-01-01:
I am getting the same error even while using Modelsim 21.2  version. Kindly help me out.