pulp-platform/axi #330: Pipelines between demux and mux would still cause deadlock in an axi_xbar?

lionheart117, 2024-02-01:
Hi,
    According to the description from axi-0.39.1/doc/axi_axbar.md as showing below, the pipelines between demux and mux would still cause deadlock in an axi_xbar. But in your paper, it says this issue have been solved. Actually I don't know which one I should believe?

> The fourth criterion is thus the only one that can be broken to prevent deadlocks. However, inserting a spill register between a master port of the demultiplexer and a slave port of the multiplexer can lead to a circular dependency inside the W FIFOs. This comes from the particular way the round robin arbiter from the AW channel in the multiplexer defines its priorities. It is constructed in a way by giving each of its slave ports an increasing priority and then comparing pairwise down till a winner is chosen. When the winner gets transferred, the priority state is advanced by one position, preventing starvation.
> 
> The problem can be shown with an example. Assume an arbitration tree with 10 inputs. Two requests want to be served in the same clock cycle. The one with the higher priority wins and the priority state advances. In the next cycle again the same two inputs have a request waiting. Again it is possible that the same port as last time wins as the priority shifted only one position further. This can lead in conjunction with the other arbitration trees in the other muxes of the crossbar to the circular dependencies inside the FIFOs. Removing the spill register between the demultiplexer and multiplexer forces the switching decision into the W FIFOs in the same clock cycle. This leads to a strict ordering of the switching decision, thus preventing the circular wait.

And the description the metioned above, I can't figure out why this situation would cause a deadlock? Is there any body could explain it in more details? Many Thanks~

micprog, 2024-02-01:
I believe the write FIFO was replaced with a credit counter in #253 (original development in #116). If I am not mistaken, the issue arose when there was uneven stalling downstream from the xbar. Unfortunately, I don't believe the doc was updated along with this fix.

lionheart117, 2024-02-02:
> I believe the write FIFO was replaced with a credit counter in #253 (original development in #116). If I am not mistaken, the issue arose when there was uneven stalling downstream from the xbar. Unfortunately, I don't believe the doc was updated along with this fix.

I guess the lockstep between AW and W might have solved this problem, but the doc was not updated. And some discription from this doc is quite obscure........