pulp-platform/axi #384: Types are very unclear

0BAB1, 2025-07-15:
Hello,

I am working on a non-industrial project and am trying to use your AXI-Lite crossbar. I'm having some difficulties understanding the types used in your implementation, specifically axi_req_t and axi_resp_t. These types are not clearly defined in include/axi/typedef.svh, at least not in a straightforward manner. They seem to be defined through an unclear macro AXI_TYPEDEF_REQ_T(full_req_t, ...) which is used in the crossbar file itself. However, full_req_t doesn't appear to be used as the crossbar input type for AXI, which adds to the confusion.

I'm trying to understand this because my AXI interfaces are implemented using SystemVerilog interface syntax. To integrate your crossbar, I need to clearly convert my interfaces to your AXI types using custom assignments. Without clear type definitions, this task is quite challenging.

Additionally, I noticed that type definitions are passed as crossbar parameters:

Copier
module axi_lite_xbar #(
  parameter axi_pkg::xbar_cfg_t Cfg = '0,
  parameter type  aw_chan_t = logic,
  parameter type   w_chan_t = logic,
  parameter type   b_chan_t = logic,
  parameter type  ar_chan_t = logic,
  parameter type   r_chan_t = logic,
  parameter type  axi_req_t = logic, // here
  parameter type axi_resp_t = logic, // here
  parameter type     rule_t = axi_pkg::xbar_rule_64_t,
  // DEPENDENT PARAMETERS, DO NOT OVERWRITE!
  parameter int unsigned MstIdxWidth = (Cfg.NoMstPorts > 32'd1) ? $clog2(Cfg.NoMstPorts) : 32'd1
) (
  input  logic                                        clk_i,
  input  logic                                        rst_ni,
  input  logic                                        test_i,
  input  axi_req_t  [Cfg.NoSlvPorts-1:0]              slv_ports_req_i,
  output axi_resp_t [Cfg.NoSlvPorts-1:0]              slv_ports_resp_o,
  output axi_req_t  [Cfg.NoMstPorts-1:0]              mst_ports_req_o,
  input  axi_resp_t [Cfg.NoMstPorts-1:0]              mst_ports_resp_i,
  input  rule_t [Cfg.NoAddrRules-1:0]                 addr_map_i,
  input  logic  [Cfg.NoSlvPorts-1:0]                  en_default_mst_port_i,
  input  logic  [Cfg.NoSlvPorts-1:0][MstIdxWidth-1:0] default_mst_port_i
);
Honestly, I'm not sure how to even declare this crossbar at this point. Could someone provide a link to a user guide or any documentation that would clarify this? I need to translate my interface into your AXI types and actually be able to use the crossbar. The configuration part was straightforward, but understanding the types is proving to be quite tedious.

I hope my questions make sense and don't come across as rude or dumb. I'm just quite lost and hoping someone can help out.

Thank you & have a good one,
BRH

zarubaf, 2025-07-15:
The AXI Lite crossbar also comes with an interface wrapper: https://github.com/pulp-platform/axi/blob/5e4bf6cf0c82ec36959b047dc75723fe8711997b/src/axi_lite_xbar.sv#L242

That should answer all your questions on the other items as well I guess.

0BAB1, 2025-07-15:
NOTA:

I was able to find the definition here:

`define AXI_TYPEDEF_REQ_T(req_t, aw_chan_t, w_chan_t, ar_chan_t) \
  typedef struct packed { \
    aw_chan_t aw; \
    logic aw_valid; \
    w_chan_t w; \
    logic w_valid; \
    logic b_ready; \
    ar_chan_t ar; \
    logic ar_valid; \
    logic r_ready; \
  } req_t;
Even though the actual type name is not the same, I had to search for the macro.

I will close this issue, but if you agree, I can add a few lines to the documentation that take the time to explain the basic types used in this repository. If yes, please specify which document I can work on for this kind of addition. Maybe the main README?

Have a good one.

0BAB1, 2025-07-15:
> The AXI Lite crossbar also comes with an interface wrapper:
> 
> [axi/src/axi_lite_xbar.sv](https://github.com/pulp-platform/axi/blob/5e4bf6cf0c82ec36959b047dc75723fe8711997b/src/axi_lite_xbar.sv#L242)
> 
> Line 242 in [5e4bf6c](/pulp-platform/axi/commit/5e4bf6cf0c82ec36959b047dc75723fe8711997b)
> 
>  module axi_lite_xbar_intf #( 
> That should answer all your questions on the other items as well I guess.

Oh okay I see thanks !

I have to say I did not have the reflex to immediately scroll down to the bottom of the file to make sens of the top inputs. This will definitely help.

Have a good one.

closed 2025-07-15 by 0BAB1